exelban/stats #1943: Network activity control

jeromepacman, 2024-05-12:
Hi
It seems that on "network" module, the activity threshold no longer works.
It appeared on the previous version (before today's)

MBP14 M2 - Sonoma 14.4.1

exelban, 2024-05-12:
Hi. I have checked that right now and it works. Could you recheck please?

Nicopalameji, 2024-05-14:
I've also noticed that the activity threshold no longer works. 

But i think the issue is that the widget is not accurately displaying the graph of the network traffic. 

The graph in the panel seems to be accurate. 

[image]
[image]

MBP14 M2 - Sonoma 14.3.1

jeromepacman, 2024-05-14:
> Hi. I have checked that right now and it works. Could you recheck please?

I have rechecked and also flushed the prefs file.
There's currently a threshold

[image]

exelban, 2024-05-14:
oh. This is a threshold for widgets only.

closed 2024-05-14 by exelban