exelban/stats #1950: When the network connection is off, latency is present and changing

Oleg-Chashko, 2024-05-17:
macOS: v12.7.5 (21H1222)
Stats: v2.10.13
MacBook Pro: (Retina, 13-inch, Early 2015)

When the network connection is off, latency is present and changing.

https://github.com/exelban/stats/assets/62497891/33045a43-7c75-4e1a-84c8-d41d2816f73e

exelban, 2024-05-19:
hmm, will take a look at that. It needs to be stopped when the network disappears.

Oleg-Chashko, 2024-05-19:
Question: For example, when pinging 192.168.1.1, why is there a difference in readings?

[image]

https://github.com/exelban/stats/assets/62497891/06f3ee91-ece1-4e48-a5bd-e4d3d3eef8f6

exelban, 2024-05-19:
I'm just providing an option for ICMP. I have no response for the server you use.

closed 2024-05-26 by exelban