exelban/stats #1955: In Sensors, unable to change fans speed.

woodymoo, 2024-05-20:
**Describe the bug**
In Sensors, unable to change fans speed. 

**Details:**
 - Device: [Macbook Pro m1 max 2021]
 - macOS: [14.5]
 - Application version: [2.10.14]

In Sensors, unable to change fans speed. 
Choose Manual, drag the indicator to the right, maximize the fan speed, but actually, the fan speeds don't change. 
You can not hear the fan spin, and temperature does not change.

closed 2024-05-20 by exelban

varalgit, 2024-05-20:
You can solve this issue with uninstalling and reinstalling the fan-helper.

woodymoo, 2024-05-22:
I only have seen there is "uninstall fan helper" in settings, but there is no reinstalling fan helper . how to reinstall?

varalgit, 2024-05-22:
If you click on the 'Sensors' menu item, you can install the fan-helper.