exelban/stats #1968: Homebrew update error: stats 2.10.14 -> 2.10.15

Ender-Wang, 2024-05-26:
**Describe the bug**
Not sure what is going on with this brew tap update, but got this, also tried removing that file and try again, same error.

**Details:**
 - Device: MacBook Pro 14 2023
 - macOS: 14.5
 - Application version: 2.10.14

```
==> Upgrading 1 outdated package:
stats 2.10.14 -> 2.10.15
==> Upgrading stats
==> Downloading https://github.com/exelban/stats/releases/download/v2.10.15/Stats.dmg
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/189285554/31081c54-f50c-450b-a870-44ca1d79e
###################################################################################################################################### 100.0%
==> Purging files for version 2.10.15 of Cask stats
Error: stats: SHA256 mismatch
Expected: 761b8546645806a881e7ade245301ea5257953621fbf43a95e95883ac72f6f62
  Actual: aba1504031e26fbf6bd29c0d962d3c7d97f35a5f1ecded5233554d42503674ca
    File: /Users/user/Library/Caches/Homebrew/downloads/8f040f5dcdcc989dbe4edab24e99b08f14402d5c46ad9bb66d459ac50988341e--Stats.dmg
To retry an incomplete download, remove the file above.
```

exelban, 2024-05-26:
looks like some problems with gh
Today I have a problem with update as weel(

Ender-Wang, 2024-05-26:
Okay, let's see how it goes in a few days, mind me, it's the first time I have this lmao

miccal, 2024-05-26:
https://github.com/Homebrew/homebrew-cask/pull/174951

closed 2024-05-26 by exelban