exelban/stats #1969: Crash on sleep

gorans, 2024-05-27:
I noticed couple of times that Stats is no longer active after waking from (deep) sleep. Checked logs and there were several crash logs that I think correspondent with entering sleep. I attached the last one.

MacOS 12.7.5
Stats 2.10.15
[stats crash log.txt](https://github.com/exelban/stats/files/15453209/stats.crash.log.txt)

exelban, 2024-05-27:
It's a bug in macOS. You need to disable monospaced font or static size for the Stack widget if it's enabled.

gorans, 2024-05-27:
I don't have Stack widget. I also don't have monospaced font or static width enabled in any widget.
I had combined modules enabled. I disabled it now and will monitor if anything changes.

exelban, 2024-05-27:
please share what the widgets look like. Because the crash report says that it crashes when trying to show that widget.

gorans, 2024-05-27:
Here are all widgets that I use:
[image]
[image]
[image]
[image]
[image]

exelban, 2024-05-27:
thx, it's a sensors widget. Please share the setting that you have for that widget in the Sensors module.

gorans, 2024-05-27:
There You go:
[image]

exelban, 2024-05-28:
hmm, interesting. Will check that.

gorans, 2024-05-28:
Just for test, I turned on "Combined modules" back on and had another crash.

exelban, 2024-05-28:
So if combined modules is off there are no crashes?

gorans, 2024-05-28:
It seems so.. :)

exelban, 2024-05-28:
hmm, strange

ryadom, 2024-05-28:
Same for me. I also using combined module.

[Stats-2024-05-27-153617.txt](https://github.com/exelban/stats/files/15467812/Stats-2024-05-27-153617.txt)

svkirillov, 2024-05-30:
I have the same issue. I use GPU, CPU, RAM, Sensors (stack, defaults), and Network modules. Modules are combined, details are not combined.

[stats-crash-log.txt](https://github.com/exelban/stats/files/15496959/stats-crash-log.txt)

exelban, 2024-06-03:
Please try this build

[Stats.dmg.zip](https://github.com/user-attachments/files/15532941/Stats.dmg.zip)

gorans, 2024-06-03:
[crash.txt.zip](https://github.com/user-attachments/files/15533695/crash.txt.zip)

exelban, 2024-06-03:
thx. Please try this one:

[Stats.dmg.zip](https://github.com/user-attachments/files/15536463/Stats.dmg.zip)

gorans, 2024-06-03:
So far so good. :)

exelban, 2024-06-03:
ok, please try this one:

[Stats.dmg.zip](https://github.com/user-attachments/files/15540051/Stats.dmg.zip)

gorans, 2024-06-05:
No crashes since v2.10.15, latest from here.
Thanks!

exelban, 2024-06-05:
hmm, interesting.

exelban, 2024-06-05:
try this one please:

[Stats.dmg.zip](https://github.com/user-attachments/files/15594336/Stats.dmg.zip)

kuba2kopec, 2024-06-05:
I've recently noticed the same issue. In my case, Stats was installed via homebrew, and today, after I found this thread, I overwrote it with the build you sent above. I'll monitor the situation and let you know if it crashes again :)

exelban, 2024-06-08:
hi. thx.
Please download the release version of the Stats, and enable the static width option in the sensors widget.

gorans, 2024-06-09:
Crash with release version.
[crash.txt.zip](https://github.com/user-attachments/files/15751397/crash.txt.zip)

exelban, 2024-06-09:
do you have enabled static width?

gorans, 2024-06-09:
Yes.

exelban, 2024-06-10:
could you please run the latest release?
and share the report, I do not have links for the build you shared crash from(

indirection42, 2024-06-11:
I guess I had the same issue these days. I have tried the latest release, but the icons still not show up in the menu bar. The weird thing is there are no crash logs in Console.app.

gorans, 2024-06-11:
Another crash with newest version:
[crash.txt.zip](https://github.com/user-attachments/files/15783963/crash.txt.zip)

kuba2kopec, 2024-06-11:
Just had a crash with the latest build in this thread:
[Stats-2024-06-10-230434.txt.zip](https://github.com/user-attachments/files/15785108/Stats-2024-06-10-230434.txt.zip)

exelban, 2024-06-11:
thx, I cannot understand why it crashes.

Could someone please try to do the following:
- disable all modules except CPU
- check if there is a crash
- enable the next module
- check if there is a crash
... end repeat that for each next module till a crash appears.

gorans, 2024-06-11:
I'll try to do that but it'll take time as I don't know what triggers the crash.
Do You want any fixed settings besides "static width" in widgets?

exelban, 2024-06-11:
the problem is here:

[image]

So when static width is enabled there must be no crash, because this line must not be triggered. But if it does not help I need to know what calling that line is. I have no idea what it could be since it's a widget drawing.

kuba2kopec, 2024-06-13:
That's interesting, because for me the crash was happening with the following widgets enabled:
[image]
I don't see the "static width" setting in any of these. Is it possible that the problematic code still triggered?

kuba2kopec, 2024-06-16:
Had another crash. I don't think that anything other than Stats version (now it's 2.10.17) changed. I haven't disabled any modules yet
[Stats-2024-06-16-170900.txt](https://github.com/user-attachments/files/15858867/Stats-2024-06-16-170900.txt)

kuba2kopec, 2024-06-16:
Actually I found something interesting. That's how the stack widget looks for me:
[image]
I'm not sure why that is

exelban, 2024-06-16:
it's because you don't have any sensors enabled

kuba2kopec, 2024-06-16:
oh, right. Weirdly I don't see the option to enable any in the settings. Not big deal for me since I don't use this widget, but I found this a bit interesting.

What I did was to enable static width and then hide the widget again. My thinking is that now, even though hidden, the static width is enabled so maybe the problematic code won't trigger

kuba2kopec, 2024-06-18:
So it didn't help, another crash happened:

[Stats-2024-06-17-173900.txt](https://github.com/user-attachments/files/15883554/Stats-2024-06-17-173900.txt)

exelban, 2024-06-18:
@kuba2kopec even with only CPU module enabled?

kuba2kopec, 2024-06-18:
I'll check that, the last crash was with CPU, GPU, sensors and network. But I can't figure out what exactly triggers the crash - sometimes my computer will go to sleep and nothing crashes

kuba2kopec, 2024-06-18:
Now I've disabled everything except CPU and we'll see what happens

svkirillov, 2024-07-04:
Got another crash:
[stats_crash.txt](https://github.com/user-attachments/files/16091840/stats_crash.txt)

ryadom, 2024-07-04:
No more crashes with:
CPU, GPU, RAM, Battery modules.
Combined modules enabled with no spacing.
Version: 2.10.17
I will keep adding more modules.

kuba2kopec, 2024-07-06:
For me no crashes up to this point with CPU, GPU and Network modules.

Seems like either I'm very lucky or the Sensors module was causing the issue. I'll wait a bit longer and then finally add the sensors module

exelban, 2024-07-06:
@kuba2kopec thx for the feedback

gorans, 2024-07-06:
I had no crashes with 2.10.18. Just updated to .19.
I'm using CPU, RAM, sensors and network widgets, combined.

I consider this bug solved. Thanks. :)

Best regards.

kuba2kopec, 2024-07-08:
I just added the sensors widget in 2.11.1. That means I'm back to my original setup of widgets.

Hope @gorans is right saying that the bug is solved. If I have another crash I'll of course let you know ;)

ryadom, 2024-07-08:
I've added all widgets (except bluetooth and clock) a few days ago. No more crashes. It seems resolved for me completely. Version: 2.10.17
Thank you!

exelban, 2024-07-08:
so only confirmation from @gorans left

gorans, 2024-07-08:
> I had no crashes with 2.10.18. Just updated to .19. I'm using CPU, RAM, sensors and network widgets, combined.
> 
> I consider this bug solved. Thanks. :)

I suppose You missed it. :)

Best regards.

kuba2kopec, 2024-07-08:
For me the sensors widget was added just 4 hours ago, so I wouldn't consider it confirmed yet, but I hope I won't have another crash since all of you are saying it shouldn't be an issue anymore

exelban, 2024-07-08:
@gorans oh, yeap. Missed that, thx.
@kuba2kopec yeap, ping me in case it will repeat.

kuba2kopec, 2024-07-08:
@exelban literally had another crash after writing the above message :/ 

I saw it now because I was AFK

[Stats-2024-07-08-204103.txt](https://github.com/user-attachments/files/16133535/Stats-2024-07-08-204103.txt)

kuba2kopec, 2024-07-09:
@exelban And another one :(

Seems like the issue isn't fixed

[Stats-2024-07-09-160817.txt](https://github.com/user-attachments/files/16146875/Stats-2024-07-09-160817.txt)

exelban, 2024-07-09:
@kuba2kopec it's not the same problem as it was originally

kuba2kopec, 2024-07-09:
Oh, okay. So what seems to be the issue now and can I help to solve it?

kuba2kopec, 2024-07-12:
Now the app seems to be crashing way more often, regardless of the enabled modules. I've had 5 crashes in the last 2 days. Do you want me to submit all of the reports?

exelban, 2024-07-13:
for now let's keep this issue, so please post the crash reports here if there are different than first one

kuba2kopec, 2024-07-15:
Sure, I don't really know what differences to look for in the reports, so here are the reports that I had since last week.

[reports_11_12_07.zip](https://github.com/user-attachments/files/16237600/reports_11_12_07.zip)

I also disabled everything except CPU and will enable them one-by-one to see if maybe I can pinpoint the issue to one module

[14 more replies]

closed 2024-11-11 by exelban