exelban/stats #1971: [Feature request] In RAM top processes show the website name rather than Safari Web Content

PuzzledUser, 2024-05-29:
Activity Monitor usually helpfully shows the website hostname in its Memory listings. This makes it easy to identify that it's _https://www.linkedin.com_ eating my memory. 

It would be great if Stats could extract the hostname, so I don't need to swap to Activity Monitor to find the culprit.

Thanks
[image]

exelban, 2024-06-24:
nope, it's not possible, because there is no this information from the `ps` output.

PuzzledUser, 2024-06-24:
Shame. I see https://apple.stackexchange.com/posts/381297/revisions talks of getting this info, but sort of fizzles on specifics.

PuzzledUser, 2024-11-16:
From the PIDs of the "Safari Web Content" processes, you can call getProcessCommandLine(pid:) for the PID to retrieve its command line arguments, which typically include the URL being accessed.

This snippet unnecessarily scans all web content processes, but illustrates the technique. 

```
import AppKit

func listSafariWebContentProcesses() {
    let runningApps = NSWorkspace.shared.runningApplications
    
    for app in runningApps {
        if app.bundleIdentifier == "com.apple.Safari" {
            let safariProcesses = NSRunningApplication.runningApplications(withBundleIdentifier: "com.apple.Safari.WebContent")
            for safariProcess in safariProcesses {
                let pid = safariProcess.processIdentifier
                if let commandLine = getProcessCommandLine(pid: pid) {
                    print("Process ID: \(pid), Command Line: \(commandLine)")
                }
            }
        }
    }
}

listSafariWebContentProcesses()
```
I hope that appeals as an approach to show this info. Thanks

More info in this Perplexity thread: https://www.perplexity.ai/search/how-does-activity-monitor-in-m-qubThwR7TEiyI9CyhK48Pw