exelban/stats #1979: "is charging" shows "no" while charging

alexander-potemkin, 2024-06-04:
**Describe the bug**
My MacBook is charging, app shows voltage, but shows 'is charging' == 'no'.
That is an old bug, I've just got my hands on reporting it.

**Details:**
 - Device: MacBook Air M1
 - macOS: 14.5
 - Application version: 2.10.11

[image]

exelban, 2024-06-05:
Don't see any problem here. 8W doesn't look like the battery is charging.

closed 2024-06-05 by exelban

alexander-potemkin, 2024-06-05:
@exelban , it's quite sufficient to charge notebook for a few hours thought... And shouldn't it be charging, if input != 0?

exelban, 2024-06-06:
I don't know, it's macOS decides. But it looks ok for me.

alexander-potemkin, 2024-06-06:
Ok, let's raise the bar up then :)

[image]

alexander-potemkin, 2024-06-06:
To rephrase - I have never seen this flag set to 'Yes'. However my Macbook has been definitely charging...

exelban, 2024-06-06:
hmm, need to check if apple does not change their API. Because this value is from macOS.

alexander-potemkin, 2024-06-06:
It seems like API might be alright - below is a screenshot from System Information app.

That's interesting... 

[image]

exelban, 2024-06-06:
but you are looking at the app, not api

And according to that screenshot it's not charging)

exelban, 2024-06-06:
I have checked on my MacBook and everything works fine.

alexander-potemkin, 2024-06-06:
> but you are looking at the app, not api

I believe the app uses API; does it not?

> And according to that screenshot it's not charging)

My point exactly! It's launched with 100W cable connected at that moment.

> I have checked on my MacBook and everything works fine.

Are you on Intel? Or it's M* laptop?

msulmanzig, 2024-06-10:
hi @alexander-potemkin can you help me in this?
https://github.com/element-hq/element-android/issues/7761

alexander-potemkin, 2024-06-18:
@exelban , it seems like this API call / flag means that the battery is charging _now_. Below is a screenshot from the same laptop that is charging.

I believe it might be beneficial to rephrase the 'Is charging' to 'Now charging', just for such narrow cases of curiousity, like mine :)

Thanks anyway for the fantastic app and your support!

[image]