exelban/stats #1988: Icons disappear from menu bar, restarting did not fix the problem.

aletei3, 2024-06-10:
Today I switched my MacBook on and the stats icons in the menu bar disappeared after a few minutes, I've already checked the settings and tried restarting both the app and my computer.
MacBook Pro 14 2021 MacOS Sonoma 14.5.0
Stats app version 2.10.16 no updates available.

exelban, 2024-06-10:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

aletei3, 2024-06-10:
Process:               Stats [1062]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.10.16 (568)
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [1059]
User ID:               501

Date/Time:             2024-06-10 10:00:26.0761 +0200
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        63FA69B8-8E9D-6D1F-BCFB-8FFD23E7B14D

Time Awake Since Boot: 240 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   Stats [1062]

Application Specific Information:
abort() called

Let me know if you need additional information, as I don't feel comfortable sharing the full stack here

exelban, 2024-06-10:
this data does not contains any useful information

aletei3, 2024-06-10:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [1062]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.10.16 (568)
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [1059]
User ID:               501

Date/Time:             2024-06-10 10:00:26.0761 +0200
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        63FA69B8-8E9D-6D1F-BCFB-8FFD23E7B14D

Time Awake Since Boot: 240 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   Stats [1062]

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x191ab2a60 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x191aeac20 pthread_kill + 288
2   libsystem_c.dylib             	       0x1919f7a30 abort + 180
3   libc++abi.dylib               	       0x191aa1d08 abort_message + 132
4   libc++abi.dylib               	       0x191a91fc0 demangling_terminate_handler() + 348
5   libobjc.A.dylib               	       0x1917301d0 _objc_terminate() + 144
6   libc++abi.dylib               	       0x191aa10cc std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x191aa4348 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib               	       0x191aa428c __cxa_throw + 308
9   libobjc.A.dylib               	       0x1917268f0 objc_exception_throw + 420
10  CoreFoundation                	       0x191b56800 -[__NSPlaceholderDictionary initWithObjects:forKeys:count:] + 728
11  CoreFoundation                	       0x191b84e18 -[NSDictionary initWithDictionary:copyItems:] + 424
12  CoreFoundation                	       0x191b84bac -[__NSPlaceholderDictionary initWithDictionary:copyItems:] + 176
13  CoreText                      	       0x1940521dc TAttributes::ApplyFont(__CFDictionary const*, __CTFont const*, TCharStream const*, CFRange, void const**) + 316
14  CoreText                      	       0x194051f2c TAttributes::TAttributes(__CFDictionary const*, TCharStream const*, CFRange, bool) + 116
15  CoreText                      	       0x193fac3d0 TTypesetterAttrString::Initialize(__CFAttributedString const*) + 576
16  CoreText                      	       0x193fabfec TTypesetterAttrString::TTypesetterAttrString(__CFAttributedString const*, __CFDictionary const*) + 132
17  CoreText                      	       0x193fabec8 CTLineCreateWithAttributedString + 140
18  UIFoundation                  	       0x1967c1bd0 __NSCoreTypesetterCreateBaseLineFromAttributedString + 432
19  UIFoundation                  	       0x1967bf000 -[NSCoreTypesetter _stringDrawingCoreTextEngineWithOriginalString:rect:padding:graphicsContext:forceClipping:attributes:stringDrawingOptions:drawingContext:stringDrawingInterface:] + 2128
20  UIFoundation                  	       0x196732f84 __NSStringDrawingEngine + 1804
21  UIFoundation                  	       0x1967745f8 -[NSString(NSExtendedStringDrawing) boundingRectWithSize:options:attributes:context:] + 124
22  UIFoundation                  	       0x19677456c -[NSString(NSStringDrawing) sizeWithAttributes:] + 40
23  Kit                           	       0x100fdc4cc 0x100f28000 + 738508
24  Kit                           	       0x100fd6a54 0x100f28000 + 715348
25  Kit                           	       0x100fd6d04 0x100f28000 + 716036
26  AppKit                        	       0x1955417d4 _NSViewDrawRect + 124
27  AppKit                        	       0x195ed2cf0 -[NSView _recursive:displayRectIgnoringOpacity:inContext:stopAtLayerBackedViews:] + 1088
28  AppKit                        	       0x1955411d8 -[NSView(NSLayerKitGlue) _drawViewBackingLayer:inContext:drawingHandler:] + 556
29  AppKit                        	       0x195a75f80 -[NSViewBackingLayer drawInContext:] + 56
30  AppKit                        	       0x1957de444 0x1953ee000 + 4129860
31  AppKit                        	       0x1958008ec 0x1953ee000 + 4270316
32  AppKit                        	       0x1957ae840 0x1953ee000 + 3934272
33  AppKit                        	       0x1958004e8 0x1953ee000 + 4269288
34  AppKit                        	       0x1957fde84 0x1953ee000 + 4259460
35  AppKit                        	       0x1957de0c8 0x1953ee000 + 4128968
36  AppKit                        	       0x1957de4a4 0x1953ee000 + 4129956
[... 529 lines cut ...]
      },
      "deploymentId" : 240000413
    },
    {
      "rolloutId" : "645eb1d0417dab722a215927",
      "factorPackIds" : {

      },
      "deploymentId" : 240000002
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000003
    },
    {
      "treatmentId" : "3dff9c91-a8fb-424e-a656-c8d6e6037574",
      "experimentId" : "662152ede2d11d1408c4db33",
      "deploymentId" : 400000015
    }
  ]
}
}

exelban, 2024-06-10:
please, use a file

aletei3, 2024-06-10:
[Stats-2024-06-10-100027-2.txt](https://github.com/user-attachments/files/15764393/Stats-2024-06-10-100027-2.txt)

aletei3, 2024-06-10:
I also tried uninstalling and reinstalling the app but it keeps crashing

exelban, 2024-06-10:
could you please share the proper crash report?
not json

moridaffy, 2024-06-10:
@exelban hi there! for some reason i also began receiving unexpected random crashes of Stats. few days ago everything was working fine, I haven't updated my system os version or anything since then. here's latest crash report :)

[Stats-2024-06-10-142605.ips.zip](https://github.com/user-attachments/files/15764885/Stats-2024-06-10-142605.ips.zip)

aletei3, 2024-06-10:
> could you please share the proper crash report? not json

This is the only crash report available on the Console Crash Report page, it is a .isp file which I had to convert to txt in order to be able to post it here. Despite this, the content is still the same.

exelban, 2024-06-10:
I have revoked the latest release, looks like #1978 causing the crash on some systems.
It's a known bug, but I hope Apple has fixed that(

I will deploy a new version today with a temporary fix.

aletei3, 2024-06-10:
> Before releasing a fix please try this build to ensure that it helps.
> 
> [Stats.dmg.zip](https://github.com/user-attachments/files/15776587/Stats.dmg.zip)

I downloaded this version 30 minutes ago and it still has to crash. Thank you very much for your support.

closed 2024-06-10 by exelban