exelban/stats #1989: When using the "Fixed" scale option in the network pop-up chart, the chart labels keep changing based on the highest visible instead of remaining fixed to the maximum axis value.

victor-marino, 2024-06-10:
**Describe the bug**
When using the "Fixed" scale option in the network pop-up chart, the chart labels are still changing dynamically based on the highest value visible in the chart, as if the scale was dynamic. They should remain fixed at the maximum axis value, as set by us in the settings.

**Details:**
 - Device: Macbook Pro M1 2020
 - macOS: 14.5
 - Application version: 2.10.16

See the screenshots below:

1. When both "peaks" from a speed test are still visible on the chart, the labels show the max speed that was achieved:
[image]

2. When the download portion of the speed test leaves the chart, the label for download speed changes back to 1.4 Mbps (next highest peak):
[image]

3. When the upload portion of the speed test leaves the chart, upload speed label also goes back to a few Kbps:
[image]

The expectation is that the chart labels should always reflect the maximum value of the axis, which in this case should be static and equal to the value we have set in the fixed scale settings (in the screenshots above, it should be 75 MBps).

exelban, 2024-07-11:
Hi. Sorry but don't see any issues here. The fixed scale allows to define the max scale of the chart. And it's not dynamic, it's a static value. On your screenshots everything looks ok for me.

closed 2024-07-11 by exelban

victor-marino, 2024-07-12:
Hi.

Sorry, I think you didn't get my point.

The actual chart (the lines) are correct. They retain the fixed scale set in the options.

The AXIS LABELS, however, are bugged. They keep changing even though they has been set to a static value. It looks like you updated the behaviour of the chart, fixing its max scale, but you missed the axis labels.

Here I have set the fixed scale to 75 MB/s:
[image]

If I make a speedtest, the scale of the chart is correct, but the AXIS LABELS set themselves to 51 MB/s and 11.3 MB/s, which is not 75 MB/s:
[image]

Here's the chart after the speedtest traffic has left the 3-minute window:
[image]

Now the AXIS LABELS are showing 133 KB/s and 12.2 MB/s, again different from the fixed 75 MB/s value I chose.

It seems like the values keep changing as if the chart was dynamic, adapting to the visible traffic, even though the scale is now static.

Can you please reopen this so the bug doesn't get lost?

exelban, 2024-07-12:
I still don't see any issues here.
Looks like the peak of the first chart for upload was 51.9 and download was 11.3.
On the second 133 and 12.2.

It's a wrong value or what?

victor-marino, 2024-07-12:
Sorry but I'm not following.

Aren't the values on the left supposed to show the scale of the Y axis? So if the scale of the Y axis is fixed, why would those values change?

When the chart scale is dynamic, it makes sense - the values keep changing to match the chart. But now the chart scale is not changing, so why do the axis labels change? I've never seen a traffic meter where the Y axis labels don't follow the Y axis scale?

victor-marino, 2024-07-12:
Ok... seems a bit strange to me (never seen it done like this before). But if it's a design decision then it's obviously not a bug and can be closed 👍🏻

exelban, 2024-07-12:
it's just a helper for me for a quick look of what was the max/min peak

victor-marino, 2024-07-12:
Yeah not totally opposed, maybe it'll grow on me overtime and I'll come to like it more than the fixed label. I'll give it some time 😬