exelban/stats #1990: Keeps crashing

noel-yap, 2024-06-10:
Stats Version: 2.10.16

Crashes started happening after rebooting this morning.

MacBook Pro
Chip: Apple M1 Max
macOS: 14.5 (23F79)

If someone can let me know how to gather other info to help troubleshoot this issue, I'll do that.

exelban, 2024-06-10:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

noel-yap, 2024-06-10:
[Stats.ips.txt](https://github.com/user-attachments/files/15776478/Stats.ips.txt)

noel-yap, 2024-06-10:
This could be the same issue as https://github.com/exelban/stats/issues/1988 but I'm able to restart sometimes but it'll crash again some time later

exelban, 2024-06-10:
thx. Looks like #1978 broke something. Yes, its the same issue.