exelban/stats #1999: Higher than normal CPU usage from system processes while Stats is running and waking from sleep

gabemaroz, 2024-06-15:
M2 MacBook Air / Sonoma 14.5

**nearbyd** / **bluetoothd** / **sharingd** All show high levels of continuous in CPU usage after waking from sleep.

With Stats off:
[image]

With Stat on:
[image]

This is with no changes in any other software. One second to the next. The high CPU usage remains as such until Stats is terminated.

exelban, 2024-06-15:
did you try to disable the bluetooth module?

gabemaroz, 2024-06-16:
Drops all of the above down to 0%. However, turning it back on brings it right back up to 30%+ for those processes. Only a restart fixes it. I've also seen some other odd behavior post-sleep. I suspect this is a bug in Sonoma 14.5 unrelated to Stats. Will file a report with Apple. Thanks.

closed 2024-06-16 by gabemaroz

fdslk, 2024-07-06:
1111

fdslk, 2024-07-06:
> 1111

2222

9840380, 2025-01-06:
same Monterey 12.7.3 bluetoothd 5% when stats work