exelban/stats #2003: [Feature Request] Prefix text for clock widget

GiorgioMendieta, 2024-06-18:
I would like to be able to prefix some text before the clock, ex '🇳🇴' so that the widget shows e.g. "🇳🇴 16:58". 

This would be really useful for people who need a quick glance at another timezone on the menubar without needing a separate app.

[image]

exelban, 2024-06-19:
[image]

You can do that already)

closed 2024-06-19 by exelban

GiorgioMendieta, 2024-06-19:
My bad, I was not aware you could add text other than hh:mm:ss on the format. Thank you!