exelban/stats #2005: Fan control no longer works

RobaloAmigo, 2024-06-22:
**Describe the bug**
I updated the program from an older version, about 5-6 versions back (I don't remember exactly). I noticed that the fan speed control stopped working. When I click on the maximum fan speed icon, nothing happens, both for the CPU and GPU fans. Manual fan speed control also is not working. Is it possible to fix this? It's very useful for me, Iuse this function for heavier tasks, where the Apple fans only start working faster when the temperatures exceeds 100 degrees Celsius.

**Details:**
 - Device: MacBook Pro M3 Max 12/30c
 - macOS: Sonoma 14.5
 - Application version: 2.1.18

Mahtabctg, 2024-06-23:
Just uninstall and reinstall Fan Helper

RobaloAmigo, 2024-06-23:
ahh OK, thanks for the tip - it was my fault, I've disabled this app from running in background because "Serhiy Mytrovtsiy" didn't sound familiar for me. @exelban is it possible to rename this helper to something more clear like "stats-fan-helper"? Thanks in advance

exelban, 2024-06-23:
@RobaloAmigo no, it's not possible. It's the name of the developer account.

closed 2024-06-23 by exelban