exelban/stats #2006: Huge space in menu bar how to adjust

9840380, 2024-06-23:
**Describe the bug**
Space use to many space how to adjust it?

[image]

Old one with previous version
[image]

**Details:**
 - Device: Macbook Pro 2015
 - macOS: 12.7.3
 - Application version: 2.10.18

exelban, 2024-06-23:
there were no changes in sizing at least for a year.

closed 2024-06-23 by exelban

9840380, 2024-06-27:
[image]

can you try to modify space (minimize space) and investigate why it not working for you icons (cpu,gpu)

defaults -currentHost write -globalDomain NSStatusItemSpacing -int 6
defaults -currentHost write -globalDomain NSStatusItemSelectionPadding -int 12

to default
defaults -currentHost delete -globalDomain NSStatusItemSelectionPadding
defaults -currentHost delete -globalDomain NSStatusItemSpacing

F1ncyJ, 2025-02-13:
Macos 15.3.1 (24D70) ，Apple M2 Pro .
Thank you for sharing, but it has no effect on me.
Setting NSStatusItemSelectionPadding and NSStatusItemSpacing and adjusting their values has no effect.