exelban/stats #2009: SSID Name not being showed now when looking at Network information.

RORWessels, 2024-06-25:
I should start by saying that I am using macOS Sequoia, so I fully expect that this could be the issue.  However, when I look at the network panel, it now shows an error where it used to show the name (SSID) of the network I am connected too at the time.  See below...

[image]

exelban, 2024-06-25:
please reopen this issue if the problem will still present in the public release

closed 2024-06-25 by exelban

nmclarty, 2024-09-17:
Just upgraded to the public release, it is still an issue here.
[image]

fjh658, 2024-09-18:
> please reopen this issue if the problem will still present in the public release

# Retrieving Wi-Fi SSID, BSSID, and Standard on macOS (Sequoia)

To retrieve the SSID and BSSID in Sequoia, location permission is required. The following Swift code demonstrates the need for location authorization when accessing Wi-Fi details:

```swift
if self.usage.connectionType == .wifi {
    if let interface = CWWiFiClient.shared().interface(withName: self.interfaceID) {
        self.usage.wifiDetails.ssid = interface.ssid()
        self.usage.wifiDetails.bssid = interface.bssid()
    }
}
```

Since macOS requires location services to access Wi-Fi information, you must request location permissions in your app. This is typically done using `CLLocationManager`, as shown below:

```swift
public override func setup() {
    // Initialize CLLocationManager
    self.locationManager = CLLocationManager()
    self.locationManager?.delegate = self

    // Request location authorization
    self.locationManager?.requestAlwaysAuthorization()
}
```

Additionally, ensure your **Info.plist** includes the necessary location permission keys and enable location services under **Capabilities**.

On macOS, using tools like `networksetup -getairportnetwork en0` or `/System/Library/PrivateFrameworks/Apple80211.framework/Versions/Current/Resources/airport` may fail, returning:

```
You are not associated with an AirPort network.
```

Additionally, running the `airport` command returns the following warning:

```
WARNING: The airport command line tool is deprecated and will be removed in a future release.
For diagnosing Wi-Fi related issues, use the Wireless Diagnostics app or wdutil command line tool.
```

Finally, using `sudo wdutil info` may still not provide the desired details, although it can return SSID and BSSID if everything works properly:

```
SSID                 : <redacted>
BSSID                : <redacted>
```

Due to these limitations, an alternative approach using `system_profiler` can be more reliable on newer versions of macOS. The following script can be used to extract the SSID:

```bash
wifi_info=$(/usr/sbin/system_profiler SPAirPortDataType)
SSID=$(echo "$wifi_info" | sed -n "/$network_interface:/,/^$/p" | awk '/Current Network Information:/{f=1; next} f==1 && /^[[:space:]]*[^[:space:]]+:/{print $1; exit}' | sed 's/://')
```

This script retrieves the current Wi-Fi network information, extracting the SSID in a reliable way.

### Determining the Wi-Fi PHY Mode

The standard way to obtain the Wi-Fi PHY mode (`activePHYMode()`) might return `unknown` on Sequoia. A better approach is to determine the PHY mode based on the transmit rate using a custom function. Here’s how you can map the transmit rate to the appropriate Wi-Fi standard:

```swift
private func determinePHYMode(fromTransmitRate rate: Double) -> String {
    if rate <= 11 {
        return "802.11b"
    } else if rate <= 54 {
        return "802.11a/g"
    } else if rate <= 600 {
        return "802.11n"
    } else if rate <= 3466.8 {
        return "802.11ac"
    } else if rate > 3466.8 {
        return "802.11ax"
    } else {
        return "unknown"
    }
}
```

You can use this function to set the Wi-Fi standard based on the transmit rate:

```swift
self.usage.wifiDetails.standard = determinePHYMode(fromTransmitRate: interface.transmitRate())
```

This approach ensures the correct Wi-Fi standard is identified, based on the data rate, even when `activePHYMode()` returns `unknown`.

exelban, 2024-09-18:
@nmclarty @fjh658 please create a new issue for that. Not mess in the closed one.

CarterLi, 2024-09-18:
Better to use `/usr/sbin/system_profiler SPAirPortDataType -xml` as it generates structured result and can be parsed directly into an NSArray