exelban/stats #2018: Dictation Feature Resets Manual Fan Settings to Auto

Remorpher, 2024-07-02:
Amazing work, fabulous app. Love at first sight, thank you very much!

I noticed an issue with the Dictation feature, a native function of macOS, that affects the manual fan settings in Stats. When using Dictation, the manual fan settings are reset to auto, but Stats does not reflect this change. Consequently, I have to manually reselect the desired fan speed to return to manual control.

DevLiuSir, 2024-07-03:
> Amazing work, fabulous app. Love at first sight, thank you very much!
>
> I noticed an issue with the Dictation feature, a native function of macOS, that affects the manual fan settings in Stats. When [...]

I also have this problem with m1max 14.5, but I never noticed it.

Exadith, 2024-07-09:
I have the impression that the switches stopped working when changing the fan operating mode. A few versions back, when I pressed the snow icon, the fans in my Mac Mini were audible, and now virtually nothing happens. When I move the slider, nothing happens either. I'm using the app version 2.11.1 on Mac Mini M2 Pro

vlbosch, 2024-07-11:
On the newer Apple Silicon Macs (M3 Max for example), the control is also reset to auto after the hardware has disabled the controlling over fan speed. So for example, when the fans are turned on by the system and I manually change the speed to fixed 3000rpm, and the systems turns the fans off again, next time they're turned on the speed is regulated by the system, yet the toolbar-item still reflects the "manual speed". Can this be edited so Stats keeps controlling the speed, even after the hardware has disabled them again?

DevLiuSir, 2024-07-11:
> On the newer Apple Silicon Macs (M3 Max for example), the control is also reset to auto after the hardware has disabled the controlling over fan speed. So for example, when the fans are turned on by [...]

On the machine of the m3 chip, it should be that Apple has changed the data. If the temperature is low enough, the hardware itself may turn off the fan to improve energy efficiency and reduce noise. In this case, the use of smc in third-party apps cannot be controlled. It can only be controlled when the system fan starts to turn. Almost all about the application of fans. It is impossible to do this.

vlbosch, 2024-07-16:
> On the machine of the m3 chip, it should be that Apple has changed the data. If the temperature is low enough, the hardware itself may turn off the fan to improve energy efficiency and reduce noise. [...]

I know that Apple determines if and when the fans turn on, and other apps can only control them when they're turned on. What I meant is the following:

-The fans are turned on, so I can control them with Stats or any given app.
-I manually set the fan speed to 55% (or any given speed)
-The fans run at 55%, until they're turned off

What then happens with Stats, is next time the hardware turns the fans on, the revert to auto, instead of the 55% I set them to. While, for example with TG-Pro, my settings are reapplied. Either a custom boost, either a set fan speed. Stats menubar-item does still present 55% and is still set to "manual", although the system (re)controls the fan speed. I have to manually set to to auto and then manual again, to regain control.

DevLiuSir, 2024-07-16:
> I know that Apple determines if and when the fans turn on, and other apps can only control them when they're turned on. What I meant is the following:
>
> -The fans are turned on, so I can control them [...]

This requires the author to answer you

closed 2024-08-12 by exelban