exelban/stats #2024: CPU always shows 0%

bdepaco67, 2024-07-07:
Hello
CPU always shows 0%
I have a MacBook Pro 13´´ 2019 Intel 7 CPU
Thanks
Best Regards
Bruno.

exelban, 2024-07-07:
where?

bdepaco67, 2024-07-07:
In the widget on top of the screen

acrostich, 2024-07-08:
This happens also on an Intel MBA. The widgets do not appear to get any update. Clicking on the widget the pane that opens up does show the values.

It works find on an M2 MB Pro.

So from the OP and my experience it looks likely this is an Intel only issue.

fokusuma, 2024-07-08:
[image]

bdepaco67, 2024-07-08:
with version v2.10.19 works fine so it is a problem of v2.11 version

exelban, 2024-07-08:
hi. Please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/16125195/Stats.dmg.zip)

oaken-source, 2024-07-08:
> hi. Please try this build:
> 
> [Stats.dmg.zip](https://github.com/user-attachments/files/16125195/Stats.dmg.zip)

I had the same problem on a clean install today, and this build fixes it.

oaken-source, 2024-07-08:
yes, intel based mac, 2019 model

bdepaco67, 2024-07-08:
That worked fine, thanks !!!

Best Regards

Bruno.

closed 2024-07-08 by exelban