exelban/stats #2031: Anonymous data prompt options in pt_BR are the same

SCHrodrigo, 2024-07-11:
**Describe the bug**
When prompting to send anonymous data during install, the pt_BR strings for accepting and not accepting are the same, not making it clear which one of to accept and to deny.

**Details:**
 - Device: Macbook Pro 2023 M3 Pro 14in
 - macOS: 14.5
 - Application version: 2.11.1

SCHrodrigo, 2024-07-11:
Current translation:

```
# Stats/Supporting Files/pt-BR.lproj/Localizable.strings:106
"Share anonymous telemetry data" = "Compartilhar dados de telemetria anônimos";
"Do not share anonymous telemetry data" = "Compartilhar dados de telemetria anônimos";
```

Proposed fix:
```
# Stats/Supporting Files/pt-BR.lproj/Localizable.strings:106
"Share anonymous telemetry data" = "Compartilhar dados de telemetria anônimos";
"Do not share anonymous telemetry data" = "Não compartilhar dados de telemetria anônimos";
```

exelban, 2024-07-11:
Hi. Could you please make a PR with that fix?

SCHrodrigo, 2024-07-11:
I just noticed there are some strings that are not translated. I can open another PR once I get the chance to look into them, if that would be fine.

exelban, 2024-07-11:
I will merge it in the Sunday. So you have time to make more changes to this PR.

PR #2033 (2024-07-14): Updated: pt_BR localization
Fixes the translation of the option to not shate telemetry data in pt-BR.

Fixes #2031
closed 2024-07-14 by exelban