exelban/stats #2042: Make Batter icon bigger in the menubar

Ahamed07, 2024-07-20:
[image]
make the battery icon bigger 
green one is from stats and the white one is batfi 
if you make the the widget bigger along with the percentage inside it will be much easier to read without straining my eyes

exelban, 2024-07-26:
hi. I increase the size of the battery widget slightly. It's max what I could except in the menu bar.

miss4chewN8, 2024-07-28:
Thank you @exelban, it is easier to read now.

gmcinalli, 2024-08-01:
Let me say though that before it was consistent with the default macOS icon style, while now it is visually larger.

The previous size allowed to use Stats as the default battery icon, hiding the macOS one, and this was completely transparent to the user, at least looking at the menu bar (not clicking it).

[image]

@exelban

gmcinalli, 2024-08-01:
Off-topic: @Ahamed07 what app is the one with a cup on your menu bar?

exelban, 2024-08-01:
@gmcinalli I agree with you. For now I don't know what option I prefer: a bigger icon or a more "native" way.
Maybe I will add a toggle that allows to specify the size of the icon. Still thinking about that....

exelban, 2024-08-01:
I will reopen this issue, just as a reminder to not forget about the option to change the battery widget size.

closed 2024-08-14 by exelban