exelban/stats #2044: console log flooded with corewifi messages

cbean, 2024-07-22:
console log flooded with corewifi messages

If the network option in Stats.app is enabled and on wifi, the console log gets flooded with corewifi messages:

[corewifi] END REQ [GET SSID] (pid=55608 proc=Stats service=com.apple.corewlan-xpc intf=en0 uuid=F7B13 err=1)
[corewifi] END REQ [GET BSSID] (pid=55608 proc=Stats service=com.apple.corewlan-xpc intf=en0 uuid=E9BC4 err=1)
[corewifi] END REQ [GET COUNTRY CODE] (pid=55608 proc=Stats service=com.apple.corewlan-xpc intf=en0 uuid=78139 err=1)

ps axw|grep -i 55608
55608   ??  S      2:42.91 /Applications/Stats.app/Contents/MacOS/Stats

Device:
Macbook Pro 16 M1 Max
OS:
macOS Sonoma 14.50

Stats:
Stats.app Version 2, installed via homebrew

exelban, 2024-07-22:
looks like opened issue in wrong project. I have nothing to do with corewifi.

closed 2024-07-22 by exelban

cbean, 2024-07-22:
> looks like opened issue in wrong project. I have nothing to do with corewifi.

So, why is the PID of Stats listed, then?