exelban/stats #2047: Stats app keeps crashing randomly

aurelioremonda, 2024-07-22:
Stats Version 2.11.3

- MacBook Air (M3)
- macOS Sonoma (14.4.1)
- Processor: 
Apple M3
8 cores, 8 threads
4 efficiency cores
4 performance cores
- Memory: 16GB
- Graphics: Apple M3 (10 cores)
- Model: Mac15,13
- Year: 2024

aurelioremonda, 2024-07-22:
I can upload the crash report if that helps

exelban, 2024-07-22:
yes please

B0rax, 2024-08-08:
Same seems to happen on my Intel Hackintosh on OSX 13.6.7. Stats 2.10.19
The mentioned workaround (disable combined modules) fixed it for now.

pdelges-chp, 2024-09-06:
Same here, iMac Pro. Just tried to disable "Combined modules", let's see if it helps.

[Stats-2024-09-06-123516.txt](https://github.com/user-attachments/files/16907580/Stats-2024-09-06-123516.txt)

blastoms, 2024-09-11:
Hello, I am experiencing the same bug on my 14-inch MacBook Pro with the Apple M3 Pro chip. I also use ‘Combined modules,’ and crashes occur once or twice a day.
[Stats-2024-09-11-100646.txt](https://github.com/user-attachments/files/16961280/Stats-2024-09-11-100646.txt)

ckujau, 2024-09-11:
It's crashing here (MacBook Pro M3) too...sometimes. Definitely during login, so having `Stats.app` in _Login items_ does nothing as it always crashes :(
Disabled "combined modules" now and even reset the settings to see if this helps anything. 🤞 

[Stats-2024-09-11-124513.ips.txt](https://github.com/user-attachments/files/16962498/Stats-2024-09-11-124513.ips.txt)

nicolopadovan, 2024-09-13:
Stats does not even open for me

rsalunga29, 2024-09-19:
Same thing here. Running M3 Pro on Sonoma. Will try to disable Combined Modules and see if it stops crashing.

closed 2024-11-11 by exelban