exelban/stats #2051: “Stats” would like to access data from other apps.

Lukaszpl666, 2024-07-26:
**Describe the bug**
When you open stats app you get a notfication:
“Stats” would like to access data from other apps.
and you have to Allow it every time you open the app, meaning when you start your Macbook you always have to hit Allow to make stats app work
[image]

**Details:**
 - Device: Macbook pro m1
 - macOS: Sequoia 15.0 (beta2,3. and 4)
 - Application version: Version 2.11.3

closed 2024-07-26 by exelban

lhvy, 2024-07-31:
This issue was closed, but _why_ is the popup shown? I am having the exact same issue, and some clarification would likely help ease user's minds, even if the popup cannot be avoided.

laszlo-t, 2024-08-05:
@Lukaszpl666, @lhvy, @exelban
same here on macOs 15, very annoying problem

found one workaround:

add the app here and allow the access
System Settings / Privacy & Security / Full Disk Access

Lukaszpl666, 2024-08-05:
@laszlo-t 
huge thanks! your workaround works brilliantly :)

lhvy, 2024-08-06:
> @Lukaszpl666, @lhvy, @exelban
> 
> same here on macOs 15, very annoying problem
> 
> 
> 
> found one workaround:
> 
> 
> 
> add the app here and allow the access
> 
> System Settings / Privacy & Security / Full Disk Access

I've just been clicking "Don't Allow" when prompted. Definitely not good practice to grant full disk access (or any file access) if not needed.

Hopefully when macOS 15 releases to the public, more people ask about this and we get clarification as to why the pop up occurs.

exelban, 2024-08-06:
I don't care at all about beta versions of macOS.

Lukaszpl666, 2024-08-06:
fair enough
but I don't think you understand the idea of beta releases ;/
beta releases are meant for developers of all other apps to check their software before the official is released so when the final release is out everything works, so now it is the best time to check your app 
the way you act is just are making beta from the final release , for me it is not a problem but for less techy users it can be. So sooner or later you will run into the same problem just the overall impression of stats app will be much lower bc your app will be one of jsut a few working incorrectly with macos 15 jsut bc of lazy devs

glasseyes42, 2024-09-08:
> I don't care at all about beta versions of macOS.

It's a good thing your github profile says "One day I shall be a professional programmer", at least you're already aware of your shortcomings. What an arrogant approach to a support request

bigruid, 2024-09-29:
It's time to reopen this issue

ihaettypo, 2024-09-29:
I'm on macos 15, seeting the same thing