exelban/stats #2055: Don't see time to discharge

jakuSk, 2024-07-28:
I don't see time to discharge in the menubar (even thought I set the correct one with % and time.

exelban, 2024-07-28:
Do you have discharge time in the popup view?

jakuSk, 2024-07-28:
Yes in the Pop up there is the Estimate

exelban, 2024-07-28:
so there is no time. It will show when macOS will provide it.

closed 2024-07-28 by exelban

jakuSk, 2024-08-01:
[image]
Why is the estimate in the pop up though

exelban, 2024-08-01:
oh, looks like I misunderstand you. But looking at your menu bar I don't see where it should be visible.

jakuSk, 2024-08-01:
[image]
Okey I found the culprit. Unfortunatelly I got confused with the UI. Even though you select the double widget, it's giving you an option to have only 1 stat there. You have to go to "Widgets" and than change mode to display both.