exelban/stats #2056: Statistical issues after merging all modules

0x0-00, 2024-07-29:
[image]
[image]

Sensor Modules I have only enabled the CPU temperature display, when I merge all the modules the stats panel counts the fastest fan revolutions, how can I get it to stop counting?

0x0-00, 2024-08-02:
[image]
only CPU diode virtual is enabled

0x0-00, 2024-08-03:
[image]
I'm sorry, I made a mistake. Nothing was turned on in the popup window.

0x0-00, 2024-08-03:
[image]
The popup doesn't have anything turned on, but it still counts Fastest fans

exelban, 2024-08-03:
yeap, looks like there is no Fastest fan in the popup settings

closed 2024-08-04 by exelban