exelban/stats #2066: Energy Impact

SuperLeiyan, 2024-08-06:
Love everything about Stats, but I just think it can do better in the battery efficiency department. I have update interval turned down to 5 - 10 seconds for everything and I'm still getting 5.2 for 12 hrs power use according to activity monitor. iStat Menus on the other hand only uses 1.2 in 12 hrs.

iState Menus has some bugs that I have to live with, and is just overall not as customisable compared to Stats and of course is also not free, but ultimately I had to make the switch due to it being so incredibly battery efficient.

Please look into this as I would love to come back to Stats cause it's overall a better app in my opinion.

Spec info:
2021 M1 Pro 16 inch Macbook Pro with 32GB Memory, running Sonoma 14.5.

exelban, 2024-08-06:
First of all, iStat Menus has separate demons for different tasks. So I hope you do not compare just the main app's efficiency.
If you know the way how Stats could be more efficient please tell me. From the beginning I've been trying to keep CPU usage as low as I can.

SuperLeiyan, 2024-08-06:
Sorry I'm not a MacOS developer and I didn't know how iStat Menus works under the hood. But just for clarifications, I only see iStat Menus Menubar and iStat Menus Helper in activity monitor, and they consume very little power, is neither of those daemon processes? If not then how can I track the actual power consumptions? Maybe iStat Menus isn't actually that efficient since the deamons that do the most heavy liftings are not tracked.

SuperLeiyan, 2024-08-06:
[image]
I found the daemon process, even then it still has minimal energy impact.

zeyugao, 2024-08-08:
If I turned on the network widget, I found that the energy impact would increase to 6~7. But if I only keep CPU, RAM, and Sensors, the energy impact would decrease to 0.x most of the time (I set the update interval to 5s or 10s, but the network cannot be set).

exelban, 2024-08-12:
Going to close that cause don't see any proposes of how it could be improved.
I'm doing my best to keep the energy impact as low as I can.

closed 2024-08-12 by exelban

AlvinZhu, 2024-09-21:
> If I turned on the network widget, I found that the energy impact would increase to 6~7. But if I only keep CPU, RAM, and Sensors, the energy impact would decrease to 0.x most of the time (I set the update interval to 5s or 10s, but the network cannot be set).

I also noticed that the network widget consumes much more power. The power consumption of other modules is acceptable, but also not better than iStat menus. Stats is great, I prefer Stats except for the issue of power consumption.
Maybe check the network module first. Sorry I'm not a MacOS developer either.