exelban/stats #2068: Can't control fans

ricardoyang00, 2024-08-08:
**Describe the bug**
Im not able to control the fans, when switching them from auto -> manual. Its always 0 RPM.

**Details:**
 - Device: [Macbook Pro 2023]
 - macOS: [15.0]
 - Application version: [2.11.5]

[image]

ricardoyang00, 2024-08-09:
Nope, didnt work to me. 
It happened since I upgraded the macOS. Probably something about Beta' OS. Lets wait for the official **Sequoia** version.

closed 2024-08-12 by exelban

AbdulR3hman, 2024-08-16:
Same issue here; using MacOS: `14.6.1`

ricardoyang00, 2024-08-16:
> Same issue here; using MacOS: `14.6.1` | `Apple M3 Pro 16-inch, Nov 2023`

I realised that if the fans are turned on by the system, you can control the RPM. But you can't activate the fans manually.