exelban/stats #2073: Improve App Name Displayed in “Allow in the Background” Section of Mac Login Items

aryanprince, 2024-08-13:
# Issue

In **Settings > General > Login Items > “Allow in the Background”** on macOS, Stats shows up under the developer’s name instead of as “Stats.app” or something more identifiable.

## Suggestion

Would it be possible to update to something more identifiable like “Stats” or "Stats (by Serhiy Mytrovtsiy)" instead of just the developer’s name? This would make it easier for users to recognize the app, ensuring it doesn’t look suspicious or unfamiliar in the settings (since I had to look up what this entry was and found out it was for the Stats app).

## Benefit
This small change will improve user trust and clarity, making the app more recognizable to everyone.

## Screenshots

Screenshot for reference for the Stats app's login item entry:
[image]

Another open-source app for example ([link](https://github.com/lwouis/alt-tab-macos)):
[image]

## Details

 - Device: Macbook Pro 14" (2021) - M1 Pro
 - macOS: 10.14.5
 - Application version: 2.11.5

closed 2024-08-13 by exelban

aryanprince, 2024-08-20:
@exelban If you don't mind me asking, just curious — could you share why this won’t be planned? Is there a technical or other limitation involved?

aryanprince, 2024-08-24:
@exelban Thanks for the clarification! Just out of curiosity, how do some other open-source Mac apps like [AltTab](https://github.com/lwouis/alt-tab-macos), [Hidden Bar](https://github.com/dwarvesf/hidden), and [MonitorControl](https://github.com/MonitorControl/MonitorControl) manage to have their app names displayed in this menu instead of the developer’s name (screenshots included)? Is there a specific fee, approval, or process from Apple that’s required for this? I’m asking to better understand how this works and if it’s something that could potentially be resolved in the future.

Also, could you please link the issue you mentioned that describes this? I’d love to read more about it. Thanks again!

## Screenshots

[image]
[image]
[image]

BitGrub, 2024-09-11:
I wish someone could find a solution to this. Almost everyone else figured it out

tinncdev, 2024-12-09:
Acorrding to this [stackoverflow](https://stackoverflow.com/questions/75918835/how-to-adjust-what-is-displayed-in-the-login-items-list-for-my-launch-daemon-age), "use the Service Management APIs to install the daemon" should be the solution.
I don't have enough mac dev experience to justify how complex the solution is.
This is not a big issue but a very very appreciated fix if possible. Every few months I go to setting to clean up things and usually forget what the item used for .

waltTAG, 2025-09-13:
agreed I didn't notice the name during the install - I had to research later when I saw "Serhiy Mytrovtsiy" in my background items.
If you want to keep the creds how about a hyphen "Stats - Serhiy Mytrovtsiy" if there is not a character limit.

BTW - I was going to buy the istat menu app - until I found this - I will give it a goo for a bit and certainly donate if I decide to keep using. I used to use a bunch of apps for fan control, processor usage, etc. This looks really nice so far.