exelban/stats #2080: Graph overlay in Popup is rendered incorrectly

LostInCompilation, 2024-08-16:
**Describe the bug**
The graph overlay has some rendering issues. In the disk Popup the graph overlay is seemingly rendered behind the graph but only for the `read` graph. The `write` graph overlay is rendered normally. This does not happen with the other Popups.
Also the overlay is sometimes rendered outside the viewable area and does not adjust its position correctly.
I also had some sporadic rendering issues, where the overlay seems to be rendered teared (see last screenshot). This however does not happen every time.

**Overlay rendered behind the Graph for `read` in Disk Popup**
[image]

---
**Out of viewable area**
[image]

[image]

---
**Sporadic rendering issue that look like tearing (doesn't happen always)**
[image]

**Details:**
 - Device: MacBook Pro 2021 M1 Pro
 - macOS: 14.6.1
 - Application version: 2.11.5

Lerbiq, 2024-08-19:
Not sure if this is related, but probably yes. I encountered something similar when I had an external drive connected. After clicking on the button for expanding/collapsing the drive statistics several times I get this
[image]

The process list renders under the graph which makes everything unreadable. The more you expand and collapse the disk details, the worse it gets.

[image]

At this point the buttons are gone and the whole thing is a mess.
Only way of fixing it is restarting the app completely.

**Details:**

- Device: MacBook Air M2 2022
- macOS: 15.0 Public Beta (24A5320a)
- Application version: 2.11.6

exelban, 2024-08-19:
@Lerbiq not related

closed 2024-09-08 by exelban