exelban/stats #2087: [Features] Please Dont clear / reset count total upload / download when updating

alsyundawy, 2024-08-23:
**Describe the bug**
A clear and concise description of what the bug is.

**Req Feature** 

- dont clear / reset count total upload / download when updating 
- reset count total upload / download move to other settings

**Details:**
 - Device: MacBook Pro 2018 15 Inch
 - macOS: MacOS
 - Application version: 2.xx.xx

exelban, 2024-08-25:
sorry, but I don't understand the question/ask. Could you please describe what do you mean?

alsyundawy, 2024-08-25:
its mean

[image]
 can you make posible the support on backup & restore

exelban, 2024-08-25:
what do you mean by backup/restore?

alsyundawy, 2024-08-26:
my case is
- when exit counting gone
- when reboot counting gone
- when update counting gone

posible counting total upload download export / impor or backup / restore on expoert setting?

closed 2024-09-08 by exelban