exelban/stats #2094: Manual fan control not resumed after macOS turned the fans off.

vlbosch, 2024-08-27:
**Describe the bug**
On newer Apple Silicon Macs, the fans can be turned off by the system and cannot manually be turned on. Only after the system decides the fans must be turned on, the speed can be manually overridden. 

If the fan speed is set to a certain percentage when the fans are turning, the speed will be maintained at said percentage. When the system turns the fans off, but later turns them on again, the fan speed doesn't return to the previously set limit. Instead, the system resumes control and therefore the speed is set automatically. Despite this, the menubar item shows that the speed is manually set at percentage X.

Either the control over the fan speed should be resumed (like TG Pro does for example), or the menubar item should indicate that the system is (automatically) controlling the fan speed. Personally I favor the first option, because I like to just 'set and forget' it. Also, I like to have another maximum speed due to noise in the working environment. I don't mind the occasional throttling you get in return for setting a custom maximum speed.

**Details:**
 - Device: MacBook Pro M3 Max
 - macOS: macOS 15 DP5, but previously also on macOS 14.6
 - Application version: 2.11.6

Mahtabctg, 2024-08-29:
Fan control doesn't work after restart or restart application, need to install fan control service again every time to make it work.

exelban, 2024-09-01:
Fan control is in legacy mode. It does not receive any updates or fixes. It's not dropped from the app just because in the old Macs it works pretty acceptable.

closed 2024-09-01 by exelban