exelban/stats #2095: Disks widget overlay

fackeltorch, 2024-08-27:
In my disk widget the third disk gets overwritten by Top Processes
[image]

exelban, 2024-08-27:
what macOS do you use?

fackeltorch, 2024-08-27:
Sorry I forgot:
iMac M1 2021
MacOS 14.6.1

exelban, 2024-08-27:
do you change any settings?
or it was like that when you start the Stats at first?

fackeltorch, 2024-08-27:
No I didn't change any settings.
After turning off the Top Processes all is ok.
After a restart of stats v. 2.11.7 it looks like this:
[image]

Mahtabctg, 2024-08-29:
[image]
[image]
I am using MacOS Sequoia developer beta 8. Disk widget is broken.

closed 2024-09-01 by exelban