exelban/stats #2105: Automatic update

jant009, 2024-09-03:
This is not a bug, more a feature needed

In case of using a dropped macOS version like 10.14 mojave, the application will automaticly update the version resulting in a incompatible error popup message from macOS.

I didn't find a way to disable automatic update. Can you provide a way to do it ?
Please
I know updating is important but can you let the choice to the user to do it or not

exelban, 2024-09-03:
In the settings of the Stats you can just change the updates check to never.
But basically it must receive the last supporting version based on macOS version. It will not work properly only if you receive an update via GitHub not via my server. Could you confirm that?

jant009, 2024-09-03:
I installed it from the .dmg and I have not the choice in the settings 
Today when I restart my mac it automaticly update to the new version.
[image]

exelban, 2024-09-03:
I'm not familiar with French. But I'm pretty sure that the first selection visible on the screenshot you shared is automatic updates. And I suppose it is "silent" now. So you have enabled auto updates. 

But it's very strange that it updates on start because I suppose on the small window it says that there is no updates. And it must receive no updates.
Strange.

jant009, 2024-09-03:
Oh ok my bad, it was on "silent" indeed. And yes the window said last version installed
Set it on "never" now but yes it should have not update. Anyway it should not happen again I hope. 
Thanks for the answer, I think you can close it

closed 2024-09-03 by exelban