exelban/stats #2115: Desktop widgets show "No data"

janhink, 2024-09-09:
**Describe the bug**
The Stats desktop widgets show "No data", although the app works and the stats are shown on menubar items correctly.

**Details:**
 - Device: Macbook Air M3, 16 GB RAM, 512 GB SSD
 - macOS: Sequoia 15.0
 - Application version: [e.g. 2.1.11]
 
[image]

F1248, 2024-09-16:
I'm experiencing the exact same bug for all five widgets Stats provides:
- MacBook Air M1 (MacBookAir10,1)
- macOS Sequoia 15.0 (24A335)
- Stats 2.11.10 (596)

jimmyxan, 2024-09-16:
exact same problem, 

Details:

Device: MacBook Pro M3pro, 18 GB RAM, 512 GB SSD
macOS: Sequoia 15.0
Application version: 2.11.10

CherrySuryp, 2024-09-17:
Same here, after update to a new MacOS Sequoia widgets are not working.

MacBook Pro 14" M3 8GB RAM, 512GB SSD
MacOS: Sequoia 15.0
Stats Version: 2.11.10

[image]

cosmic-latte, 2024-09-17:
The same issue here, I guess this is obviously caused by Sequoia changing some APIs.

F1248, 2024-09-17:
When launching Stats, this error appears in Console:
```
Couldn't read values in CFPrefsPlistSource<0x600001ad0a20> (Domain: eu.exelban.Stats.widgets, User: kCFPreferencesAnyUser, ByHost: Yes, Container: (null), Contents Need Refresh: Yes): Using kCFPreferencesAnyUser with a container is only allowed for System Containers, detaching from cfprefsd
```

This seems to be a known occurrence and not new to macOS Sequoia:
https://developer.apple.com/forums/thread/738322

The error seems to cause `userDefaults` to be nil here:
https://github.com/exelban/stats/blob/7a574d9530540d7d1632046291c7dcbae7a4e7bc/Modules/CPU/widget.swift#L31

Which causes `entry.value` to remain nil:
https://github.com/exelban/stats/blob/7a574d9530540d7d1632046291c7dcbae7a4e7bc/Modules/CPU/widget.swift#L43-L45

Which causes the widget to show `No data` here:
https://github.com/exelban/stats/blob/7a574d9530540d7d1632046291c7dcbae7a4e7bc/Modules/CPU/widget.swift#L62
https://github.com/exelban/stats/blob/7a574d9530540d7d1632046291c7dcbae7a4e7bc/Modules/CPU/widget.swift#L99-L101

I don't know how to fix this, but maybe this information will be helpful for someone to fix it.

LordZeuss, 2024-09-19:
I am also experiencing the same issue, how ever it also applies to the widgets in notification center.

[image]

Macbook Air 15" M3
MacOS Version Sequoia 15.0
Most up to date Stats version.

ufukcelen, 2024-09-21:
got same problem

MacBook Pro 14" M3 PRO 18GB RAM, 512GB SSD
MacOS: Sequoia 15.0
Stats Version: 2.11.10

[Image](https://github.com/user-attachments/assets/c65fc226-75ef-4ba8-8215-cbb3ef9cd7cd)

dmercalli, 2024-09-22:
+1 here 
MacBook Air 13" M2 8GB ram, 256SSD
MacOS Sequoia 15.0
Stats version 2.11.10

arthurreynoob, 2024-09-24:
Same problem with the widgets but menu bar stats works fine

Macbook Air 13" M2 24GB ram, 512SSD
MacOS Sequoia 15.0
Stats version 2.11.10

F1248, 2024-09-24:
Just out of interest, does this affect anyone on an Intel Mac?

lcsvcn, 2024-09-24:
same issue here, m1 pro 2021, mac os sequoia too

closed 2024-09-29 by exelban

CherrySuryp, 2024-09-29:
@exelban
After update "No Data" persists. Now it doesn't display "No Data" but there's no metrics in widget

MacBook Pro 14 M3 8GB
MacOS: Sequoia 15
Stats Version: 2.11.13

[image]

jagai, 2024-10-06:
@CherrySuryp This might sound really dumb, but did you try "Settings" -> "Reset settings"? This helped me fix the issue with the widgets not displaying any data.

CherrySuryp, 2024-10-07:
@jagai yeah, that helped, thanks😅

Luk1nh4, 2024-10-07:
> @CherrySuryp This might sound really dumb, but did you try "Settings" -> "Reset settings"? This helped me fix the issue with the widgets not displaying any data.

In my case I even formatted the system, but it didn't help, as I mentioned here >> https://github.com/exelban/stats/issues/2167 only the CPU widget shows "no data"

AlbatorLaho, 2025-01-09:
> Just out of interest, does this affect anyone on an Intel Mac?

Yes, I'm running Sequoia (15.2), on a MacBook 2019 with an Intel Core i9-9980HK

> @CherrySuryp This might sound really dumb, but did you try "Settings" -> "Reset settings"? This helped me fix the issue with the widgets not displaying any data.

Weird, that did seem to fix the Disk widget, but not the CPU widget...