exelban/stats #2120: Fan control suddenly stopped working

BitGrub, 2024-09-12:
Whether I change fan speed manually or by hitting the snowflake button, fans don't change, sensor seems to be reading the fan speed accurately though. Feature used to work, no idea when/why it stopped though. Tried reseting settings, completely uninstalling Stats, pressed the "Uninstall fan helper" button in the settings, nothing changed. Please help

Macbook M1 Max, macOS Sonoma 14.6.1

BitGrub, 2024-09-12:
Also, I deleted the background task, in /Library/LaunchDaemons (or ~/Library/LaunchAgents, I don't remember), and it won't come back no matter how many times I reinstall the app and reset/import settings

exelban, 2024-09-13:
please read the FAQ.

closed 2024-09-13 by exelban