exelban/stats #2122: Stats crashes on startup

nicolopadovan, 2024-09-13:
**Describe the bug**
When opening the app - in any way and any circumstances - it fails to do so, producing a crash report.
This is a reopening of #2121 since I previously though there was no crash report.
This issue might be related to #2047 

Full Crash Report can be found [here](https://gist.github.com/nicolopadovan/840971bdd4047e605d4b2060f4a93dce)
Translated Crash Report [here](https://gist.github.com/nicolopadovan/80def2156dbd6f3f4f09e0731c8e09f1)

Console produces this related error:
```
Couldn't read values in CFPrefsPlistSource<0x6000015b0ea0> (Domain: eu.exelban.Stats.widgets, User: kCFPreferencesAnyUser, ByHost: Yes, Container: (null), Contents Need Refresh: Yes): Using kCFPreferencesAnyUser with a container is only allowed for System Containers, detaching from cfprefsd
```

**Details:**
 - Device: MacBook Pro 2021 16" / M1 Max / 64GB RAM
 - macOS: Sonoma / 14.6.1
 - Application version: 2.11.9 / tested down to 2.11.7

@exelban let me know if I should have continued on the other thread or if I am right opening this new one. Not trying to bother or create any inconvenience :D 
I am available for further debugging if needed

exelban, 2024-09-13:
Thx, please use the latest version of the Stats. Does it crash at the start or after some time?
And did it work previously?

nicolopadovan, 2024-09-13:
I am using the latest version, [you can find all the reports for the 2.11.9 version here](https://gist.github.com/nicolopadovan/c316160f0f4475094f2c4d4b80fbb0b5).

It crashes right away. The icon does not appear anywhere at all, there is no sign of life.
The only thing that happened - but I can't consistantly reproduce - is a rapid flash of the widgets; they disappear right after. This has appened 2/3 times out of the 30-ish I've tried.

It stopped working in the last couple of days. I can't be more precise due to the fact that I was away from my computer during this period of time.

exelban, 2024-09-13:
try to run this build:
[Stats.dmg.zip](https://github.com/user-attachments/files/16998754/Stats.dmg.zip)

nicolopadovan, 2024-09-13:
[Crash Report](https://gist.github.com/nicolopadovan/145201bc90dade3b4c381de4267d5f30)

exelban, 2024-09-13:
hmm, thx. What about this one?
[Stats.dmg.zip](https://github.com/user-attachments/files/16999101/Stats.dmg.zip)

PS: I'm just disabling the features till the crash stops cause don't see the crash reason(

nicolopadovan, 2024-09-13:
> hmm, thx. What about this one? [Stats.dmg.zip](https://github.com/user-attachments/files/16999101/Stats.dmg.zip)
> 
> PS: I'm just disabling the features till the crash stops cause don't see the crash reason(

Fair enough.

[Another crash report :D](https://gist.github.com/nicolopadovan/692efb0d5d3faa23a6f26614f19284f8)

nicolopadovan, 2024-09-13:
In order to help you out - and avoid you having to "waste" time on this: are you able to reproduce the error? Could it be on my end? Or it would not generate a crash report in that case?

exelban, 2024-09-13:
nope, don't care about the logs. There is nothing useful in 99% cases. Only for fine tune/debug.
What about this:

[Stats.dmg.zip](https://github.com/user-attachments/files/16999275/Stats.dmg.zip)

nicolopadovan, 2024-09-13:
This one works!

exelban, 2024-09-13:
[Stats.dmg.zip](https://github.com/user-attachments/files/16999364/Stats.dmg.zip)

nicolopadovan, 2024-09-13:
Did you manage to fix the issue? Will you be releasing a patch?

The last one produces [this Crash Report](https://gist.github.com/nicolopadovan/ea31a9df822b236dbab965bbafa09028)

nicolopadovan, 2024-09-13:
Running the app from Xcode, the error appears on `Stats/Modules/Disk/popup.swift` at `init(width:size:free:)` (line 590)

```
        self.usedBarSpace = NSView(frame: NSRect(x: 0, y: 0, width: width, height: view.frame.height))
```

```
Thread 1: EXC_BREAKPOINT (code=1, subcode=0x194121d64)
```

exelban, 2024-09-13:
hmm, looks like disk module causes the crash. Now we need to fix the problematic place.
Just to ensure that it's a disk module there is a build with disabled only the Disk module:
[Stats.dmg.zip](https://github.com/user-attachments/files/16999464/Stats.dmg.zip)

PS: wow, I was not expecting this part.

exelban, 2024-09-13:
please put: `print(width, view.frame.width, percentage)` before `self.usedBarSpace = NSView(frame: NSRect(x: 0, y: 0, width: width, height: view.frame.height))`

And try to run the app.

nicolopadovan, 2024-09-13:
Output:
```
Couldn't read values in CFPrefsPlistSource<0x600000cea5b0> (Domain: eu.exelban.Stats.widgets, User: kCFPreferencesAnyUser, ByHost: Yes, Container: (null), Contents Need Refresh: Yes): Using kCFPreferencesAnyUser with a container is only allowed for System Containers, detaching from cfprefsd
2024-09-13 23:20:07 reader.swift:85 DBG [<CPU.LoadReader: 0x151f7dbe0>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:159 DBG [<CPU.ProcessReader: 0x600000cf8c60>] Set update interval: 1 sec
2024-09-13 23:20:07 reader.swift:85 DBG [<CPU.ProcessReader: 0x600000cf8c60>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:159 DBG [<CPU.AverageReader: 0x600000cf8fc0>] Set update interval: 60 sec
2024-09-13 23:20:07 reader.swift:85 DBG [<CPU.AverageReader: 0x600000cf8fc0>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:85 DBG [<CPU.TemperatureReader: 0x600000cf9050>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:85 DBG [<GPU.InfoReader: 0x600000ce3570>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:85 DBG [<RAM.UsageReader: 0x6000001cd1d0>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:159 DBG [<RAM.ProcessReader: 0x600000ceafd0>] Set update interval: 1 sec
2024-09-13 23:20:07 reader.swift:85 DBG [<RAM.ProcessReader: 0x600000ceafd0>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:85 DBG [<Disk.CapacityReader: 0x600001098280>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:159 DBG [<Disk.ActivityReader: 0x6000010f4080>] Set update interval: 1 sec
2024-09-13 23:20:07 reader.swift:85 DBG [<Disk.ActivityReader: 0x6000010f4080>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:159 DBG [<Disk.ProcessReader: 0x600000cf0240>] Set update interval: 1 sec
2024-09-13 23:20:07 reader.swift:85 DBG [<Disk.ProcessReader: 0x600000cf0240>] Successfully initialize reader
2024-09-13 23:20:07 reader.swift:85 DBG [<Sensors.SensorsReader: 0x600000fc83c0>] Successfully initialize reader
2024-09-13 23:20:08 readers.swift:288 DBG [<Sensors.SensorsReader: 0x600000fc83c0>] Found 2 fans
Requested print on next line
208.78265233945257 246.0 0.8487099688595633
Requested print on next line
nan 246.0 nan
```

exelban, 2024-09-13:
wow. Please put `print(size, free, size)` before `let percentage = CGFloat(size - free) / CGFloat(size)`

nicolopadovan, 2024-09-13:
Code:
```
    public init(width: CGFloat, size: Int64, free: Int64) {
        self.size = size
        
        super.init(frame: NSRect(x: 0, y: 0, width: width, height: 10))
        
        let view: NSView = NSView(frame: NSRect(x: 1, y: 0, width: self.frame.width - 2, height: self.frame.height))
        view.wantsLayer = true
        view.layer?.borderColor = NSColor.secondaryLabelColor.cgColor
        view.layer?.borderWidth = 0.25
        view.layer?.cornerRadius = 3
        self.background = view
        
        print("Size, Free, Size")
        print(size, free, size)
        let percentage = CGFloat(size - free) / CGFloat(size)
        let width: CGFloat = (view.frame.width * (percentage < 0 ? 0 : percentage)) / 1
        print("Width, view.frame.width, percentage")
        print(width, view.frame.width, percentage)
        self.usedBarSpace = NSView(frame: NSRect(x: 0, y: 0, width: width, height: view.frame.height))
        self.usedBarSpace?.wantsLayer = true
        self.usedBarSpace?.layer?.backgroundColor = NSColor.controlAccentColor.cgColor
        
        view.addSubview(self.usedBarSpace!)
        self.addSubview(view)
        
        self.widthAnchor.constraint(equalToConstant: self.frame.width).isActive = true
        self.heightAnchor.constraint(equalToConstant: self.frame.height).isActive = true
    }
```

Output:
```
Couldn't read values in CFPrefsPlistSource<0x60000142a910> (Domain: eu.exelban.Stats.widgets, User: kCFPreferencesAnyUser, ByHost: Yes, Container: (null), Contents Need Refresh: Yes): Using kCFPreferencesAnyUser with a container is only allowed for System Containers, detaching from cfprefsd
2024-09-13 23:21:45 reader.swift:85 DBG [<CPU.LoadReader: 0x12dfafae0>] Successfully initialize reader
2024-09-13 23:21:45 reader.swift:159 DBG [<CPU.ProcessReader: 0x60000143d5f0>] Set update interval: 1 sec
2024-09-13 23:21:45 reader.swift:85 DBG [<CPU.ProcessReader: 0x60000143d5f0>] Successfully initialize reader
2024-09-13 23:21:45 reader.swift:159 DBG [<CPU.AverageReader: 0x60000143d950>] Set update interval: 60 sec
2024-09-13 23:21:45 reader.swift:85 DBG [<CPU.AverageReader: 0x60000143d950>] Successfully initialize reader
2024-09-13 23:21:45 reader.swift:85 DBG [<CPU.TemperatureReader: 0x60000143d9e0>] Successfully initialize reader
2024-09-13 23:21:45 reader.swift:85 DBG [<GPU.InfoReader: 0x600001422520>] Successfully initialize reader
2024-09-13 23:21:46 reader.swift:85 DBG [<RAM.UsageReader: 0x60000190e580>] Successfully initialize reader
2024-09-13 23:21:46 reader.swift:159 DBG [<RAM.ProcessReader: 0x60000143e250>] Set update interval: 1 sec
2024-09-13 23:21:46 reader.swift:85 DBG [<RAM.ProcessReader: 0x60000143e250>] Successfully initialize reader
2024-09-13 23:21:46 reader.swift:85 DBG [<Disk.CapacityReader: 0x60000086f580>] Successfully initialize reader
2024-09-13 23:21:46 reader.swift:159 DBG [<Disk.ActivityReader: 0x600000854900>] Set update interval: 1 sec
2024-09-13 23:21:46 reader.swift:85 DBG [<Disk.ActivityReader: 0x600000854900>] Successfully initialize reader
2024-09-13 23:21:46 reader.swift:159 DBG [<Disk.ProcessReader: 0x600001409320>] Set update interval: 1 sec
2024-09-13 23:21:46 reader.swift:85 DBG [<Disk.ProcessReader: 0x600001409320>] Successfully initialize reader
2024-09-13 23:21:46 reader.swift:85 DBG [<Sensors.SensorsReader: 0x600001710fa0>] Successfully initialize reader
2024-09-13 23:21:46 readers.swift:288 DBG [<Sensors.SensorsReader: 0x600001710fa0>] Found 2 fans
Size, Free, Size
994662584320 150482533356 994662584320
Width, view.frame.width, percentage
208.78265233945257 246.0 0.8487099688595633
Size, Free, Size
0 0 0
Width, view.frame.width, percentage
nan 246.0 nan
```

exelban, 2024-09-13:
wtf. Looks like some of your disk have 0 - size and free. Strange.

nicolopadovan, 2024-09-13:
Very weird, I will check that as well; glad I've found an edge case for Stats though :D

nicolopadovan, 2024-09-13:
By the way, running `diskutil list`

```
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *1.0 TB     disk0
   1:             Apple_APFS_ISC Container disk1         524.3 MB   disk0s1
   2:                 Apple_APFS Container disk3         994.7 GB   disk0s2
   3:        Apple_APFS_Recovery Container disk2         5.4 GB     disk0s3

/dev/disk3 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +994.7 GB   disk3
                                 Physical Store disk0s2
   1:                APFS Volume Macintosh HD - Data     886.1 GB   disk3s1
   2:                APFS Volume Macintosh HD            10.3 GB    disk3s3
   3:              APFS Snapshot com.apple.os.update-... 10.3 GB    disk3s3s1
   4:                APFS Volume Preboot                 6.2 GB     disk3s4
   5:                APFS Volume Recovery                941.6 MB   disk3s5
   6:                APFS Volume VM                      20.5 KB    disk3s6

/dev/disk4 (disk image):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        +17.6 GB    disk4
   1:                 Apple_APFS Container disk5         17.6 GB    disk4s1

/dev/disk5 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +17.6 GB    disk5
                                 Physical Store disk4s1
   1:                APFS Volume iOS 17.5 21F79 Simul... 17.0 GB    disk5s1

/dev/disk6 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *4.0 TB     disk6
   1:                        EFI EFI                     209.7 MB   disk6s1
   2:                  Apple_HFS LaCie                   4.0 TB     disk6s2

/dev/disk7 (disk image):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        +40.9 MB    disk7
   1:                  Apple_HFS Stats                   40.9 MB    disk7s1

/dev/disk8 (disk image):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        +40.9 MB    disk8
   1:                  Apple_HFS Stats                   40.9 MB    disk8s1

/dev/disk9 (disk image):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        +40.9 MB    disk9
   1:                  Apple_HFS Stats                   40.9 MB    disk9s1

/dev/disk10 (disk image):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        +40.9 MB    disk10
   1:                  Apple_HFS Stats                   40.9 MB    disk10s1
```

exelban, 2024-09-13:
please try this build:
[Stats.dmg.zip](https://github.com/user-attachments/files/16999539/Stats.dmg.zip)

nicolopadovan, 2024-09-13:
Works correctly; only shows two disks (MHD and LaCie)

exelban, 2024-09-13:
thx.
I will make a debug build with some logs gathering to get more information what disk caused the problem. If you have time to run it I will be appreciative, cause I'm very interested in what it was.

nicolopadovan, 2024-09-13:
Surely! I am available

nicolopadovan, 2024-09-13:
At this point, I am wondering whether this is an issue with Stats, or whether there is some kind of "ghost" disk in my machine

nicolopadovan, 2024-09-14:
After the night, during which I turned off the mac, the error represents itself. [Crash Report](https://gist.github.com/nicolopadovan/53d18ad1adef7cbe128a4730d1a20353)

nicolopadovan, 2024-09-14:
One point of failure might be here:

```
    private func freeDiskSpaceInBytes(_ path: URL) -> Int64 {
        do {
            if let url = URL(string: path.absoluteString) {
                let values = try url.resourceValues(forKeys: [.volumeAvailableCapacityForImportantUsageKey])
                if let capacity = values.volumeAvailableCapacityForImportantUsage, capacity != 0 {
                    return capacity
                }
            }
        } catch let err {
            error("error retrieving free space #1: \(err.localizedDescription)", log: self.log)
        }
        
        do {
            let systemAttributes = try FileManager.default.attributesOfFileSystem(forPath: path.path)
            if let freeSpace = (systemAttributes[FileAttributeKey.systemFreeSize] as? NSNumber)?.int64Value {
                return freeSpace
            }
        } catch let err {
            error("error retrieving free space: \(err.localizedDescription)", log: self.log)
        }
        
        return 0
    }
    
    private func totalDiskSpaceInBytes(_ path: URL) -> Int64 {
        do {
            let systemAttributes = try FileManager.default.attributesOfFileSystem(forPath: path.path)
            if let totalSpace = (systemAttributes[FileAttributeKey.systemSize] as? NSNumber)?.int64Value {
                return totalSpace
            }
        } catch let err {
            error("error retrieving total space: \(err.localizedDescription)", log: self.log)
        }
        
        return 0
    }
```

Where, on error, you just return 0 for both the total disk space and the free disk space

exelban, 2024-09-14:
Looks like some problems with the disk driver or the disk itself. I have never seen something like this.

nicolopadovan, 2024-09-14:
I am currently running some diagnostics on the LaCie disk, brb

nicolopadovan, 2024-09-14:
First Aid is taing far more than expected (it's a 4TB HDD working on USB-C). I'll post the results here as soon as available (probably tomorrow morning)

nicolopadovan, 2024-09-14:
@exelban I can confirm that the disk does NOT appear to have any issues

exelban, 2024-09-15:
so maybe drivers

closed 2024-09-15 by exelban

nicolopadovan, 2024-09-15:
I did a quick health checkup on the drivers as well, but there isn't anything out of the ordinary.
I will continue trying to inspect the issue, using a fork of Stats, and if I manage to find the issue **within** the package I will post a PR. 
If I find that the issue is on my machine, I'll let you know here.
Thanks for the help so far! :D

exelban, 2024-09-15:
thx for your feedback and quick help)