exelban/stats #2125: [Feature Request] Clock module: Support dynamic time zones (to auto-switch daylight saving time)

hengginator, 2024-09-15:
**Describe the bug/idea**
To align alternative time zones in the clock module closer to reality (actually used time), it would be great to support dynamic daylight saving zone configuration (to auto-reflect actually used time for certain area/city). 
As time zones are currently based on direct "UTC offset" only, it cannot auto-switch with summer/winter time change (daylight saving time) for a certain desired clock entry.

A very simple way of improving this would be allowing time zone config to be based on local time offset (ie: "localTime -2hrs" instead of only "UTC -2hrs"). For many time zones (ie: on the same continent usually) this would work fine as daylight saving times zone often change together.

For a global "full version" of such a feature a full "city time zone chart" resp. an zone index including information of winter/summer time start/end dates would be perfect since as daylight saving times are starting/ending on different dates globally.

**Details:**
 - Device: [e.g. Macbook Pro 2016]
 - macOS: [e.g. 10.15.5]
 - Application version: [e.g. 2.1.11]

exelban, 2024-10-03:
hi, thanks for the feedback.
I will take a look what could be done here.

hengginator, 2024-10-21:
....time zone data could be sourced from [TimezoneDB](https://timezonedb.com) which offers offline data to download & implement, as well as a RestAPI  (free even for commercial use under [Creative Commons Attribution 3.0 License](https://creativecommons.org/licenses/by/3.0/)). 

Alternatively there is also [WorldTimeAPI](http://worldtimeapi.org).

hengginator, 2025-03-26:
FTR: seem to be addressed by #2447

exelban, 2025-03-27:
yeap, that fixes this "problem"

closed 2025-07-06 by exelban