exelban/stats #2127: Clock name not updated in Popup for last entry

hengginator, 2024-09-15:
**Describe the bug**
Changed clock name string is not updated in popup (nor in popup config tab) for last (added) clock entry.

Workaround: 
Add one more (unused) clock. This will update name of previously last clock. Then new (unused) clock entry can be deleted again. This way last clock name is updated as configured.

**Details:**
 - Device: [e.g. Macbook Pro 2016]
 - macOS: [e.g. 10.15.5]
 - Application version: [e.g. 2.1.11]

AntonioAEMartins, 2024-09-18:
Hi @hengginator,

I've tested on my computer and encountered the same issue. I'm downloading the repository to my local machine now and will dig into it to find the problem.

Details:

- Device: MacBook Pro 2019 (i9)
- macOS: Sequoia

I'll keep you updated as I investigate further.

AntonioAEMartins, 2024-09-18:
Hi @exelban,

What's the best way to assign myself to this issue? Are there any contribution guidelines or rules I should follow before proceeding?

closed 2024-10-06 by exelban