exelban/stats #2131: (macOS Sequoia) Icons in the menu bar are not grayed out on the inactive screen.

au-ganymede, 2024-09-17:
**Bug:**
As in the topic. On an inactive (second) monitor, the Stats icons in the menu bar are not grayed out, unlike the other icons, thus drawing unnecessary attention to themselves.

---

**Details:**
 - Device: **Macbook Pro M3 Max**
 - macOS: **Sequoia 15.0 (24A335)**
 - Application version: **2.11.5**

---

[image]

whusnoopy, 2024-09-18:
Same on an other MacBook Pro 16 with M3 Max, macOS 15.0 (24A335), and Stats 2.11.10

nicolopadovan, 2024-09-18:
Same on MacBook Pro 2021 M1 Max, MacOS 15.0.0, Stats 2.11.0

gmcinalli, 2024-09-18:
Confirmed also here, running Stats 2.11.9 on macOS 15.0.0.

[image]

AntonioAEMartins, 2024-09-18:
Hi @au-ganymede, @nicolopadovan, @gmcinalli, and @whusnoopy,

I'm also using Stats and encountering the same issue on my machine (MacBook Pro i9 16"). I'll be downloading the repository locally to investigate and attempt to find a solution. Have any of you already done this? If so, what did you discover?

au-ganymede, 2024-09-18:
@AntonioAEMartins 

It seems it's a bug in Sequoia. 

[From AlDente GitHub:](https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/1322#issuecomment-2355317028)

> Thank you for this bug report. We are aware of this issue and have tried to fix it already. Unfortunately, this is a bug of macOS Sequoia, and we have already filed a bug report at Apple since it is not only a problem of AlDente but most Menubar apps that use a different framework to also display more complex data in the Menubar. We hope Apple will fix it soon and will communicate updates here.

AntonioAEMartins, 2024-09-18:
Hi @au-ganymede,

Thank you for pointing that out regarding the Sequoia bug.If that’s the case, do you think we should close this issue for now and wait for Apple to fix it, or is it worth keeping open for further tracking?

exelban, 2024-09-18:
@au-ganymede thx for sharing this info

au-ganymede, 2024-09-18:
@AntonioAEMartins 
I’m not sure, I’ll leave that decision to the moderator. 
If it were up to me, though, I’d probably keep the thread open until Apple fixes the bug, following the example of the statement I mentioned earlier, to avoid duplicate reports.

@exelban 
No problem. 

By the way, thanks to the email notification of your comment, I now know exactly who's listed under the "Allow in the Background" section in System Settings 😅 
I was initially confused, expecting to see the app's name (Stats) instead of the developer's name there 😃

[image]

gmcinalli, 2024-09-18:
@au-ganymede Really strange, some other apps works normally here, for example, look at KeepingYouAwake (the coffee cup) in my screenshot.

au-ganymede, 2024-09-18:
@gmcinalli From what I understand, this bug affects apps that use non-standard frameworks (rather than the built-in ones) to display complex information in the menubar. Apps like Stats, AlDente, and LittleSnitch all fall into this category and are impacted by the issue.

exelban, 2024-09-27:
@KizzyCode thx, I will try that

exelban, 2024-10-13:
hi. Please someone check if its fixed in this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/17355853/Stats.dmg.zip)

KizzyCode, 2024-10-13:
Seems to be fixed for me 🎉

Main screen:
[image]

Secondary screen:
[image]

gmcinalli, 2024-10-14:
@exelban Resolved!

[image]

closed 2024-10-20 by exelban

KingfuChan, 2024-10-20:
Issue persists in 2.11.5. MacBook Air 13" (M2), macOS Sequoia (15.0.1)

On active screen:
[image]

On inactive screen:
[image]

fran-6co, 2024-10-20:
issue keeps happening for me on my inactive monitor

Stats 2.11.5 – MBP 13" Touchbar 2018 – Sequoia 15.0.1

[image]

whusnoopy, 2024-10-21:
> Same on an other MacBook Pro 16 with M3 Max, macOS 15.0 (24A335), and Stats 2.11.10

Fixed on MacBook Pro 16 with M3 Max, macOS 15.0.1 (24A348), and Stats 2.11.15

sassrobi, 2024-10-28:
When "Combined modules" switched on, this bug still exists. Without combined modules, it is indeed fixed. MacBook Pro 16, M3 Pro

exelban, 2024-10-28:
@sassrobi will fix it, thx