exelban/stats #2133: Import networking support for macOS Sequoia

CarterLi, 2024-09-18:
`/usr/sbin/networksetup -getairportnetwork en0` no longer works in Sequoia, which always prints `You are not associated with an AirPort network.`

The only working method I can find is [`/usr/sbin/system_profiler SPAirPortDataType -detailLevel basic -xml`](https://github.com/fastfetch-cli/fastfetch/commit/756865855b111d835f1ee1002f88ecf7b83eb5b7) which costs about 2 seconds in my MBP. Let me know if someone finds better methods.

closed 2024-09-18 by exelban

CarterLi, 2024-09-18:
???

Corrently Wi-Fi SSID is detected as `You are not associated with an AirPort network`. You should remove the nonsense string at least @exelban

CarterLi, 2024-09-18:
Ref: https://github.com/exelban/stats/commit/dca3c03acd4050d4ee354b81741fab275c3083dc#diff-b989e3c0653d12da46746d7d294a22de7830cfbe67e24d984feb2c880547aca9R355

exelban, 2024-09-18:
please take a look at #2009

CarterLi, 2024-09-18:
Ok. Didnt find tha as it was closed

exelban, 2024-09-18:
yeap, I was thinking that it's a new issue(