exelban/stats #2134: [Bug] Fans cannot be started manually

guzkrzysztof, 2024-09-18:
**Describe the bug**
I can't manually start fans after macOS update to Sequoia 15.0

**Details**
- Device: MacBook Pro M3 Pro
- macOS: Sequoia 15.0
- Application version: 2.11.10

[image]

closed 2024-09-18 by exelban

noormohammedb, 2024-09-23:
after updating into macOS 15.0 (Sequoia) I cant mannualy adjust fan speed from `stats`

guzkrzysztof, 2024-09-23:
@exelban can we get some explanation why this was marked as not planned?
I think the fan control is one of the best features in the `stats`, so it is really sad that it stopped working.

noormohammedb, 2024-09-23:
He mentioned It does not receive any updates or fixes [#2094](https://github.com/exelban/stats/issues/2094#issuecomment-2323394880)

guzkrzysztof, 2024-09-23:
@noormohammedb thanks, nevertheless this forces users to find another tool for fan control :/