exelban/stats #2142: [enhancement] Hide fan speed when it's 0?

Ender-Wang, 2024-09-22:
Just found out that most of the time in winter, my fan is off, and only under heavy workloads they turn on, so would it be possible to show the fan speed numbers only when fans are on? Screenshots: the two 0s on the right are the fan speed indicator. [image]
**Details:**
 - Device: Macbook Pro 2023
 - macOS: macOS 14.7
 - Application version: 2.11.10

exelban, 2024-09-23:
I don't see any reason to hide that. It requires too complicated logic to maintain the features it provides.

closed 2024-09-23 by exelban

Ender-Wang, 2024-09-23:
> I don't see any reason to hide that. It requires too complicated logic to maintain the features it provides.

Thanks, that makes sense.