exelban/stats #2143: Virtual Network Interface (utun) Being Detected as Primary Interface Instead of Physical Interface

fjh658, 2024-09-23:
# 🐞 **Describe the bug**:

When using a virtual private network (VPN) on macOS, the virtual network interface (e.g., `utun`) is being detected as the primary interface instead of the physical network interface (e.g., `en0` for Wi-Fi). This leads to incorrect network information being displayed in the app, such as "Unknown" for the interface and physical address.

This issue occurs when retrieving the primary network interface using `SCDynamicStoreCopyValue` and `State:/Network/Global/IPv4`. The virtual network interface appears at the top, resulting in the wrong interface being prioritized.

```bash
ip-info
DEV: (Hardware Port: Wi-Fi, Device: en0)
Type: Wireless
SSID: my-mobile
Interface_IPv4: 192.168.126.194
Utun_List:
- uInterface:      utun4
- uInterface_IPv4: 198.18.0.1
- uPID:            1095
- uCMD:            /Applications/Surge.app/Contents/MacOS/Surge
DNS_Server(en0): 8.8.8.8 1.1.1.1
DNS_Server(utun4): 198.18.0.2
```

```bash
echo "show State:/Network/Global/IPv4" | scutil
<dictionary> {
  PrimaryInterface : utun4
  PrimaryService : 32A30073-9328-4F0F-BF03-D63367E18584
  Router : 198.18.0.1
}
```

```bash
scutil --nwi
Network information

IPv4 network interface information
   utun4 : flags      : 0x5 (IPv4,DNS)
           address    : 198.18.0.1
           VPN server : 254.1.1.1
           reach      : 0x00000003 (Reachable,Transient Connection)
     en0 : flags      : 0x5 (IPv4,DNS)
           address    : 192.168.126.194
           reach      : 0x00000002 (Reachable)

   REACH : flags 0x00000003 (Reachable,Transient Connection)

IPv6 network interface information
   No IPv6 states found

   REACH : flags 0x00000007 (Reachable,Transient Connection,Connection Required)

Network interfaces: utun4 en0
```

📸 **Screenshot**:

[image]

---

### 🚶‍♂️ **Steps to Reproduce**:

1. 🖥️ Connect to a VPN that creates a virtual network interface (e.g., `utun4`).
2. 💻 Run the application that uses `SCDynamicStoreCopyValue(nil, "State:/Network/Global/IPv4" as CFString)` to determine the primary interface.
3. ⚠️ The app will display the virtual network interface (`utun4`) as the primary interface instead of the physical network interface (`en0`).
4. 🔍 In the network interface display, the app shows "Unknown" for the interface and physical address.

---

### 🎯 **Expected Behavior**:

The app should correctly display the physical interface (e.g., `en0`) and its associated details, such as the interface name and physical address, even when a VPN is connected.

---

### 💥 **Actual Behavior**:

The virtual network interface (`utun`) is detected as the primary interface, leading to "Unknown" being displayed for the interface and physical address in the network settings.

---

### 🛠️ **Details**:

- **Device**: 🖥️ all
- **macOS**: 🍏 all
- **Application version**: 📦 [2.11.11]

---

### 🧩 **Proposed Solution**:

Modify the logic for determining the primary interface to prioritize the physical interface over the virtual interface.

**Current Code**:

```swift
private var primaryInterface: String {   
    get {
        if let global = SCDynamicStoreCopyValue(nil, "State:/Network/Global/IPv4" as CFString), let name = global["PrimaryInterface"] as? String {
            return name
        }
        return ""
    }
}
```

**Proposed Solution**:

```swift
private var primaryInterface: String {        
    get {
        if let interfaceName = CWWiFiClient.shared().interface()?.interfaceName {
            return interfaceName
        }
        return ""
    }
}
```

This modification checks for a Wi-Fi interface first using `CWWiFiClient.shared().interface()`. If it exists, it returns the Wi-Fi interface name, ensuring that the physical interface is prioritized over any virtual interfaces.

exelban, 2024-09-23:
don't see any problems, just change the interface in the settings.

fjh658, 2024-09-23:
> don't see any problems, just change the interface in the settings.

NO，you are currently using automatic detection, and automatically detecting the virtual network interface doesn't have any practical meaning.
[image]

fjh658, 2024-09-23:
For the network status, automatic detection should normally automatically retrieve the user's default physical network interface and get the most commonly used setting, rather than requiring the user to select it manually.

exelban, 2024-09-23:
what do you mean no?
why you cannot?

fjh658, 2024-09-23:
> what do you mean no? why you cannot?

It's not that I can't do it, but it can be done better. The purpose of automatic detection is to make things easier for the user. If a specific physical network interface is manually set, when the network environment changes, does the user have to manually reset it again? You can refer to the experience of iStat Menus.