exelban/stats #2146: Performance and internet connection problems after Sequoia update

ErwannRousseau, 2024-09-24:
**Describe the bug**

Since the Sequoia update, the cpu was spiking and the internet connection was terribly reduced, I'm convinced that it was Stats that caused this problem because as soon as I removed it my connection went from ~12Mb/s to over 420Mb/s (which is what I normally have). 

**Details:**
 - Device: [e.g. Macbook Pro 13 pouces, M2, 2022]
 - macOS: [e.g. 15.0]
 - Application version: I've forgot to check before uninstall

ealmuina, 2024-09-24:
+1 to this issue. 

I have spent the whole day figuring out what was happening to both my home and work Macbooks, the wifi speed and latency were terrible. After closing Stats, they went back to normal right away.

This seems to be true only for wifi though. I tried connecting directly to the router through Ethernet, and everything works normal that way while having Stats running.

brndnsh, 2024-09-24:
I noticed issues with this today as well. I was able to work around the problem by disabling the network section of Stats.

exelban, 2024-09-24:
please take a look at the processes. There is a process that takes suspiciously too many cpu?

laszlo-t, 2024-09-24:
Same here on Intel Macbook Pro 2020
I haven't seen any suspicious cpu spikes, processes.

pings:
`ping -c 20 192.168.1.1`

average: 41 vs. 9

with Stats:

```
PING 192.168.1.1 (192.168.1.1): 56 data bytes
64 bytes from 192.168.1.1: icmp_seq=0 ttl=64 time=2.386 ms
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=10.323 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=57.509 ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=1.211 ms
64 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=40.668 ms
64 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=47.102 ms
64 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=89.782 ms
64 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=1.323 ms
64 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=31.465 ms
64 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=51.538 ms
64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=102.144 ms
64 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=2.493 ms
64 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=1.294 ms
64 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=111.009 ms
64 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=1.448 ms
64 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=9.982 ms
64 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=3.671 ms
64 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=48.340 ms
64 bytes from 192.168.1.1: icmp_seq=18 ttl=64 time=92.766 ms
64 bytes from 192.168.1.1: icmp_seq=19 ttl=64 time=121.585 ms

--- 192.168.1.1 ping statistics ---
20 packets transmitted, 20 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 1.211/41.402/121.585/40.819 ms

```

disabled network in Stats:

```
PING 192.168.1.1 (192.168.1.1): 56 data bytes
64 bytes from 192.168.1.1: icmp_seq=0 ttl=64 time=11.587 ms
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=10.901 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=11.357 ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=3.258 ms
64 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=1.706 ms
64 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=10.986 ms
64 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=3.905 ms
64 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=9.797 ms
64 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=1.822 ms
64 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=9.732 ms
64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=10.919 ms
64 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=11.522 ms
64 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=11.981 ms
64 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=9.810 ms
64 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=1.733 ms
64 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=9.972 ms
64 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=9.420 ms
64 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=10.187 ms
64 bytes from 192.168.1.1: icmp_seq=18 ttl=64 time=11.406 ms
64 bytes from 192.168.1.1: icmp_seq=19 ttl=64 time=10.180 ms

--- 192.168.1.1 ping statistics ---
20 packets transmitted, 20 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 1.706/8.609/11.981/3.631 ms
```

frg96, 2024-09-25:
Agreed. When I turn off Network in Stats, the rate goes back to normal. Otherwise, it's throttled

exelban, 2024-09-25:
hmm, try to disable ICMP in the network settings.

ealmuina, 2024-09-25:
> hmm, try to disable ICMP in the network settings.

I see no changes. This is Stats with Network module enabled and ICMP disabled:

[image]

And this is it with network disabled:

[image]

exelban, 2024-09-25:
please try to disable top process

ealmuina, 2024-09-25:
> please try to disable top process

Slightly better, but still slower than normal:

[image]

ealmuina, 2024-09-25:
> Ok, this build for fine for me. Please someone take a look to see if it fixes the problem for you as well:
> 
> [Stats.dmg.zip](https://github.com/user-attachments/files/17129507/Stats.dmg.zip)

That one works like a charm on my end too:

[image]

Thanks a lot @exelban !!!

exelban, 2024-09-25:
new release with hot fix is available. Please update to the latest release.

closed 2024-09-25 by exelban

wtchangdm, 2024-09-25:
Wow. Just saw the release note when I was scratching my head because two of my laptops show different speed
I know this issue has been closed, I am just here to say it did get fixed

Nice finding and nice fix!

Before
[image]
After
[image]

brndnsh, 2024-09-25:
Thanks for the hotfix, it works great!