exelban/stats #2152: Apple could not verify "Stats.app" is free of malware

FelikZ, 2024-09-27:
**Describe the bug**
After auto-update I can't launch the app anymore since Apple can't verify is geniune:
[image]

I am using [homebrew cask](https://github.com/Homebrew/homebrew-cask/blob/master/Casks/s/stats.rb) to manage Stats.app.

Any ideas?

**Details:**
 - Device: [e.g. Macbook Pro 2021 / M1]
 - macOS: [e.g. 15.1 BETA]
 - Application version: [e.g. 2.1.12]

exelban, 2024-09-27:
I do not maintain the app in the brew, so I cannot help with that.
You can download the latest release [here](https://github.com/exelban/stats/releases/tag/v2.11.12).

closed 2024-09-27 by exelban

MyDogTom, 2025-10-06:
I confirm that downloading Stats directly from release page solves the problem.

Homebrew has an issue about this: https://github.com/Homebrew/homebrew-cask/issues/165584
Unfortunately, the error seems to be sporadic and hard to investigate. That's why Homebrew issue is also closed as `not planned`.