exelban/stats #2154: Getting this popup each time: "Stats" would like to access data from other apps.

ihaettypo, 2024-09-29:
**Describe the bug**
Each time Stats is launched there is a popup: "Stats" would like to access data from other apps.

**Details:**
 - Device: Macbook Pro 2021
 - macOS: 15.0
 - Application version: 2.11.12

CollOfTheWild, 2024-09-29:
This popup is extremely annoying. I wanted to comment here just in case people come looking, this was not a decision by the dev and not his fault. This is to do with stricter permissions in macOS Sequoia, and as far as I know there is no way to disable it.

Someone please find a workaround!!

exelban, 2024-09-29:
It's strange, cause I've never seen this type of popup. Will take a look at what is related to.

CollOfTheWild, 2024-09-29:
I'll take a screenshot when I get back to my machine.

It's related to privacy updates in macOS 15. Every 3rd party app that reads nearly any kind of data forced the user to "Allow" the app to read their data on every boot and I believe once per week without a reboot.

It's very annoying for me because my UPS is being repaired and I have very unstable electricity and have to shut my machine down whenever I'm not using it. So I have to click allow on Stats multiple times a day.

Really can't believe they would put in a policy so anti 3rd party. I get the intention, protecting user data, but this is way too far in my opinion.

exelban, 2024-09-29:
Im more curious why I don't see this popup

CollOfTheWild, 2024-09-29:
[Here](https://cyberinsider.com/apple-announces-tightened-security-measures-in-macos-sequoia/) is an article about it. The notification is similar to this one, except the one pictured with the "Stop" sign is only for apps that record the screen like Chrome Remote Desktop. For stats I think it is the app icon instead of the "Stop" sign.

I would try a reboot after making sure you are on macOS 15. I wonder if it may be region dependent too.

CollOfTheWild, 2024-09-29:
[image]

jaiakash, 2024-09-30:
Yeah even i have this popup for every log in or restart. Can we somehow allow it from settings so we don't get popup everytime?

Ca-Wi, 2024-10-01:
Same here, every login or reboot.

harryqt, 2024-10-02:
Yeah, every time I reboot it pops up.

exelban, 2024-10-03:
looks like its some [bug](https://forums.developer.apple.com/forums/thread/758375) in the new macOS related to the app groups that allow communicate between app and widgets.

Still strange that I'm not able to see this window. Even on a clean Mac that has never installed Stats. Hope it will be fixed in the next release of macOS.

Im curious if it's affecting all users or only in some specific cases, maybe by SoC or something.

CollOfTheWild, 2024-10-03:
I'm on 15.1, M2 SoC.

laszlo-t, 2024-10-03:
try this
https://github.com/exelban/stats/issues/2051#issuecomment-2269876548

exelban, 2024-10-03:
@laszlo-t yes, I have seen that, but deleting, disabling or enabling does not change the situation. Also, I have noticed that you have enable that. Could you disable it and check what changed?

laszlo-t, 2024-10-03:
I'm not the good candidate to test this.
Because the problem disappeared for me some days ago with the v2.11.13 (Intel 15.1 beta). I don't need the workaround anymore.

But I can reproduce it if I go back to the previous version. In that case I can only eliminate the popup if I enable the full disk access (keeping the entry there in disabled state is not enough).

harryqt, 2024-10-05:
The problem seems to have gone away automatically for me.

Ca-Wi, 2024-10-08:
> The problem seems to have gone away automatically for me.

Same here .... :-)

exelban, 2024-10-25:
I'm interested if anyone still has a problem with that?

closed 2024-11-03 by exelban

PuzzledUser, 2024-11-08:
I haven't see that message, but today saw one about access to the LAN after updating to Mac OS 15.1 yesterday. Version 2.11.16
[image]