exelban/stats #2161: [Mac Pro 2013] Power information incorrect

nullpointerninja, 2024-10-03:
When using Stats with a Mac Pro 2013 (model macpro6,1) all the Power related information is shown as 0.00W.

Example of what I see with this machine:
[image]

I found this iSMC tool that correctly shows Power information with the same machine: https://github.com/dkorunic/iSMC.
[image]

Do you think this could be changed in Stats so that it shows the correct information like iSMC does?

 - Device: Mac Pro 2013
 - macOS: Monterey 12.7.6
 - Application version: 2.11.13

exelban, 2024-10-03:
Hi. If you would like to provide a list of sensors that should be added I will add them.
It could be in the format: name - key.

nullpointerninja, 2024-10-03:
Hello. That's one question I have. I'm not sure if it's an unmapped key or if it's something else.
For example, I see the DC In field is shown but with the value being 0.00W. Is it because the PDTR key is not mapped correctly, or is it reading the correct key but something else is amiss?

These are the main(?) keys with their respective names for this MacPro6,1 model:

```
CPU Core 1 - PC0C
CPU Package - PCPC
CPU Package Total - PCPT
CPU SA High Side - PC0S
CPU Total - PCTR
DC In - PDTR
Ext GPU 1.8V - PG1C
GPU - PG0C
GPU 1 - PG0R
GPU 2 - PG1R
GPU A VDDCI Low-Side - PG0S
GPU B VDDCI Low-Side - PG1S
HDD1 Low-Side - PH0R
Main 12V Rail - PD2R
Memory Total - PMTR
System Total - PSTR
```

exelban, 2024-10-03:
hmm, could you please share the output of: `/Applications/Stats.app/Contents/Resources/smc list | grep "PDTR"`

exelban, 2024-10-03:
it's only one correct sensors that works also on M series, so quite strange

nullpointerninja, 2024-10-03:
I do see PDTR with a 0.0 value when running that command (without piping to grep).

...
[PCTR]     0.0
[PD2R]     0.0
**[PDTR]     0.0**
[PG0C]     0.0
[PG0R]     0.0
[PG0S]     0.0
[PG1C]     0.0
[PG1R]     0.0
[PG1S]     0.0
[PGTR]     0.0
[PH0F]     129.0
...

exelban, 2024-10-03:
strange. It could be some bug in my smc module. But I don't have any intel based Mac for debug anymore (and on previous MacBooks pro) this value was ok. And it will take too much time to debug with debug builds. So if you really want to have this value you could try to check the smc from the project you share and compare outputs with mine. Maybe it's some problem with value conversion or something like that. Since it works on the newest macOS it's not a priority for me.

You could also check all sensors that you share to add. Maybe they have the same problem in Stats. And there is no sense to add them.

nullpointerninja, 2024-10-03:
Checking your SMC code I do see something that might be related. You have both "fpe2" and "fp2e" (e and 2 in both ways) but only "sp5a". From the other utility I see most of the Power related keys have a type of "spa5" (a and 5 backwards).
Could this be part of the problem?

exelban, 2024-10-03:
yes, it could be a problem. Try to run smc from this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/17248890/Stats.dmg.zip)

nullpointerninja, 2024-10-03:
Wow you're fast!
[image]

exelban, 2024-10-03:
ok, so looks like it works. I will add missing sensors from list you share in the next release.

nullpointerninja, 2024-10-03:
Missed one extra sensor, though I'm not sure what it actually is... it's:
TBT IO High Side - PI0R

Thanks for the quick help! Much appreciated!

closed 2024-10-06 by exelban