exelban/stats #2167: macOS sequoia 15.0.1 bug (widget only)

Luk1nh4, 2024-10-07:
I Have the same issue as https://github.com/exelban/stats/issues/2156 (closed), only the CPU widget show as "no data",  

**Details:**
 - Device: iMac 2019
 - macOS: 15.0.1
 - Application version: e.g. 2.11.13

[image]

exelban, 2024-10-07:
do you have enabled CPU module?

Luk1nh4, 2024-10-07:
> do you have enabled CPU module?

Yes.

exelban, 2024-10-07:
take a look here #2115

closed 2024-10-07 by exelban

CherrySuryp, 2024-10-07:
@Luk1nh4 sorry for offtop, but can you share your wallpaper? I really like how it looks like😁

exelban, 2024-10-07:
unfortunately apple makes a lot of mess in the new macOS with app groups. All widgets uses the same code base for communication so only some macOS problems could be here. Please wait for the next macOS, maybe they will fix that. At this point I cannot do anything(

Luk1nh4, 2024-10-11:
> @Luk1nh4 sorry for offtop, but can you share your wallpaper? I really like how it looks like😁

https://github.com/user-attachments/assets/f665cc20-f0b8-4772-b961-88fc707b3fe8

It's not actually a wallpaper, it's a looping video

Luk1nh4, 2024-10-11:
> unfortunately apple makes a lot of mess in the new macOS with app groups. All widgets uses the same code base for communication so only some macOS problems could be here. Please wait for the next macOS, maybe they will fix that. At this point I cannot do anything(

No problem, I understand.

CherrySuryp, 2024-10-11:
@Luk1nh4 thanks a lot