exelban/stats #2172: [Bug] Combined modules - Sensor section view bug and restart position of sensor items.

fran-6co, 2024-10-09:
When activating the combined modules, no combined details, the sensor module behaves with two issues:

- Every time the app starts, the sensor module restarts its order, and I would like to have memory for the order they appear.

- The "System Total" value in "Power" section, doesn't show the complete unit, there is missing the `h` of `Wh`. Seems partially visible when the number is not that long, attached two screenshots without the `h`.

  - When the value is not long:

     - [image]
     - [image]

  - When the value is long:

     - [image]

My settings for the combined module section:

[image]

porg, 2024-10-24:
> Every time the app starts, the sensor module restarts its order, and I would like to have memory for the order they appear.

1) Note that luckily there is a first rudimentary explicit feature to get #928. 
- I will shortly share instructions/screenshots how this can be achieved.

2) There is also an implicit one, which you and also me have observed, which is currently yet not persisted at all.
- The order in which you enable sensors for the very first time determines their order within the widget.
- But this is not persisted. Neither in the default settings. Nor when exporting and later re-importing. I will make a particular followup issue on this too.

porg, 2024-10-24:
@fran-6co reported:
> Every time the app starts, the sensor module restarts its order, and I would like to have memory for the order they appear.

I'm a fellow user and a contributing user experience designer and I have some good news on your inquiry:

1) Note that you can meanwhile achieve #928. 
    - See: [How to set the order of sensor items in the widget](https://github.com/exelban/stats/issues/928#issuecomment-2435183959)
    - It's a rudimentary first version which hopefully gets improved
    - I filed these proposals: #2193

2) There is also an implicit method to set sensor order, which you have observed, which is currently yet not persisted at all.
    - The order in which you enable sensors for the very first time determines their order within the widget.
    - But this is not persisted. Neither in the default settings. Nor when exporting and later re-importing.
    - I filed this bug: #2192

closed 2024-11-14 by exelban