exelban/stats #2173: Fan speed control doesn't synchronize anymore

LostInCompilation, 2024-10-09:
**Describe the bug**
Since the last update to 2.11.14 (or 1-2 versions earlier, can't say for sure), the synchronization of the fan speed doesn't work anymore. I tried turning the `"Synchronize fan's control"` setting off and back on, still doesn't work. When I set the speed of one fan, the other fan doesn't change its speed.

**Details:**
 - Device: MacBook Pro M1 Pro
 - macOS: 15.0.1
 - Application version: 2.11.14
 -

closed 2024-10-13 by exelban

LostInCompilation, 2024-10-13:
If you close this as not planned I assume you have a speculation or know what is causing this issue. For possible contributors it would be helpful if you can elaborate a bit.

hiromon0125, 2024-10-21:
I had this issue but I fixed it by going to the settings and clicked on "uninstall fan helper" then going to sensors tab and clicked on "install fan helper".