exelban/stats #2178: Widgets don't consistently change text colour according the background

proteino, 2024-10-10:
**Describe the bug**
I will talk about the RAM widget specifically, but this also occurs for the user widgets. Depending on the background color of the menu bar, the widgets descriptor ("RAM" in this case) changes its color, so it is readable all the time. But the value of the widget ("72%" e.g.) does not, so it matches the background color sometimes and is therefore not visible. You can see an example in the two screenshots:

value is visible: 
[image]

value is not visible:
[image]

**Details:**
 - Device: [MacBookAir M2]
 - macOS: [15.0.1]
 - Application version: [2.11.14]

exelban, 2024-10-10:
what color have you set for this widget?

proteino, 2024-10-11:
I tried multiple, for the sake of demonstration I chose black for the example screenshots. But ideally I would like black text color, which turns to white on black background (exactly like the widget descriptor "RAM" does it automatically)

exelban, 2024-10-11:
if you select black color it will be black.
If you want dynamic use monochrome

closed 2024-10-11 by exelban